openwrt/packages #2749: reaver fails to download

hnyman, 2016-05-26:
reaver package fails to download as the googlecode svn is not available any more.

Maybe an archived .tar.gz could be used instead?
https://code.google.com/archive/p/reaver-wps/downloads

Download error:
```
make[2]: Entering directory '/Openwrt/trunk/feeds/packages/net/reaver'
mkdir -p /Openwrt/trunk/dl
SHELL= flock /Openwrt/trunk/tmp/.reaver-r113.tar.gz.flock -c '	 echo "Checking out files from the svn repository..."; mkdir -p /Openwrt/trunk/tmp/dl && cd /Openwrt/trunk/tmp/dl && rm -rf reaver-r113 && [ \! -d reaver-r113 ] && ( svn help export | grep -q trust-server-cert && svn export --non-interactive --trust-server-cert -r113 http://reaver-wps.googlecode.com/svn/trunk reaver-r113 || svn export --non-interactive -r113 http://reaver-wps.googlecode.com/svn/trunk reaver-r113 ) && echo "Packing checkout..." && 	tar czf /Openwrt/trunk/tmp/dl/reaver-r113.tar.gz reaver-r113 && mv /Openwrt/trunk/tmp/dl/reaver-r113.tar.gz /Openwrt/trunk/dl/ && rm -rf reaver-r113; '
Checking out files from the svn repository...
svn: E170013: Unable to connect to a repository at URL 'http://reaver-wps.googlecode.com/svn/trunk'
svn: E160013: '/svn/trunk' path not found
svn: E170013: Unable to connect to a repository at URL 'http://reaver-wps.googlecode.com/svn/trunk'
svn: E160013: '/svn/trunk' path not found
Makefile:56: recipe for target '/Openwrt/trunk/dl/reaver-r113.tar.gz' failed
make[2]: *** [/Openwrt/trunk/dl/reaver-r113.tar.gz] Error 1
make[2]: Leaving directory '/Openwrt/trunk/feeds/packages/net/reaver'
package/Makefile:196: recipe for target 'package/feeds/packages/reaver/download' failed
```

yousong, 2016-05-26:
Thanks for informing me of this.  As this reaver tool seems to have been inactive for quite a long time.  I will check if a switch to another more recent/active fork is suitable.

yousong, 2016-05-26:
Just opened a pr to fix the broken url issue in case this might be a blocker issue.  I will wait for a while for possible review comments then merge it.

- https://github.com/openwrt/packages/pull/2750

hnyman, 2016-05-26:
Just merge it. The package does not download currently, so it is fundamentally broken :-(

yousong, 2016-05-26:
Merged.

closed 2016-05-26 by yousong