openwrt/packages #2754: python-pip binary file

linamu, 2016-05-27:
Hi,
i compile python-pip and its okey
in my "build_dir/target-mipsel_r2_uClibc-0.9.32/pip-8.1.1/ipkg-install/usr/bin" 
the binary files contains a path from my machine. is that normal ?

`file pip
pip: Python script, ASCII text executable`

`vi pip
!/home/ubuntu/openwrt/staging_dir/target-mipsel_r2_uClibc-0.9.32/usr/bin/hostpython
# EASY-INSTALL-ENTRY-SCRIPT: 'pip==8.1.1','console_scripts','pip'
__requires__ = 'pip==8.1.1'
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.exit(
        load_entry_point('pip==8.1.1', 'console_scripts', 'pip')()
    )
`

commodo, 2016-05-27:
which OpenWrt version are you using/building ?  trunk or CC/15.05 ?
and which python version (trunk or CC/15.05 ?

yousong, 2016-08-02:
I just check with python-pip-8.1.2 and the shebang looks just fine.

The issue has been open for more than 2 months without further progress, so I am going to close it.  Feel free to reopen it if there are updates.

closed 2016-08-02 by yousong