openwrt/packages #2789: nginx: Ticks all the right boxes but is missing SSL

Xotic750, 2016-05-31:
I know that OpenWRT is built with small devices in mind, but storage size is increasing and there are many that run OpenWRT on x86 machines (PCs).

I recently  went through the process of creating a such a net facing service and reviewed the options available to me. I began with uhttp as that was already installed with SSL enabled for luci, I soon found problems with certificates (signed by [Let's Encrypt](https://letsencrypt.org/)) in that it was not honouring a full walk of the certificate whatever I tried and I could only manage a B grade result with [SSL Labs](https://www.ssllabs.com/ssltest/). 

I moved on to Apache, and while I have plenty of space the bloat was just too much the wiki warned of an [unsafe configuration when using ScriptAlias for PHP](https://www.ssllabs.com/ssltest/). Enough to say that I was put off.

I moved on, nginx, this looks promising but after install I found that SSL is disabled by default. Downloaded the SDK, enabled SSL, compiled, forced it onto my box. Loaded my certificates, fixed the weak DH key, configured it for PHP fastcgi etc. and attained an A+ grade with SSL Labs. It was fast, I was able to proxy luci and znc interfaces through it, redirect all HTTP traffic to HTTPS and my certificates were being walked correctly. Perfect, look no further.

The problem is the lack of SSL out of the the box. So I have 2 suggestions.

1) Enable SSL on the nginx build as default.
This option may not be acceptable as OpenWRT attempts to cater for the small devices and in general all other packages have SSL disabled by default (znc is an exception here where SSL is enabled by default).

2) Create an nginx-ssl packages.
This is what seems to be the way it is handled by the majority of packages where SSL may be required by an end user. Examples would be luci, nmap and ncat.

So I would like to hear your thoughts about how to ease the situation. I would be willing to attempt creating a nginx-ssl if that is the way to go, perhaps there is already a mechanism in the build process to achieve this simply. I just do not know.

heil, 2016-05-31:
Hi,

Thomas Heil

> 
> I know that OpenWRT is built with small devices in mind, but storage size is increasing and there are many that run OpenWRT on x86 machines (PCs).

Well so I do but Iam not aware that they are many other.

> I recently went through the process of creating a such a net facing service and reviewed the options available to me. I began with uhttp as that was already installed with SSL enabled for luci, I [...]
Ok
> I moved on to Apache, and while I have plenty of space the bloat was just too much the wiki warned of an unsafe configuration when using ScriptAlias for PHP. Enough to say that I was put off.

Apache is a bit outdated at the moment. Please open a seperate ticket for that
> I moved on, nginx, this looks promising but after install I found that SSL is disabled by default. Downloaded the SDK, enabled SSL, compiled, forced it onto my box. Loaded my certificates, fixed the [...]
You are free to send a patch. Maybe a build variant is a way to do that.
> This option may not be acceptable as OpenWRT attempts to cater for the small devices and in general all other packages have SSL disabled by default (znc is an exception here where SSL is enabled by [...]
indeed send a pr.
> So I would like to hear your thoughts about how to ease the situation. I would be willing to attempt creating a nginx-ssl if that is the way to go, perhaps there is already a mechanism in the build process to achieve this simply. I just do not know.

give haproxy a try. it works with ssl out of the box. because haproxy is not a webserver i would suggest nginx or uhhtp for the backend

cheers 
thomas

Xotic750, 2016-05-31:
Hi Thomas,
I will open issues for uhttp and for Apache as soon as I can put a little information together.
I will take a look at haproxy, though I am very happy with nginx just now. Thanks.
I will take a look at how SSL is handled in nmap etc. and see if I can replicate the work here, though I not very experienced, my code is/will likely be verbose and will need follow up to minimise line/character counts. Expect a PR, with any luck. :)

Xotic750, 2016-06-01:
I have made an attempt at suggestion 2) https://github.com/openwrt/packages/pull/2793
Built and running on my own machine, I based the workings on the Makefile of dnsmasq. I'm sure that with knowledge, improvement can be made.

closed 2016-06-01 by Xotic750