openwrt/packages #2795: apache:

Xotic750, 2016-06-01:
I was going to trial Apache,  but the wiki warned of an [unsafe configuration when using ScriptAlias for PHP](https://wiki.openwrt.org/doc/howto/http.apache#configuring_apache_and_php5). A search found the following explanation: http://insecurety.net/?p=912 and [CVE-2012-1823](http://www.cvedetails.com/cve/CVE-2012-1823) I am not sure of when or if this has been fixed or if it relates to Apache or PHP, but the Apache package is a little out of date and bump is required. If this is indeed fixed then the wiki could use an update with some further information.

I would send a PR for the bump but I have not been able to get it to compile on my setup.

**Makefile**
```
@@ -6,19 +6,19 @@
 #
 
 include $(TOPDIR)/rules.mk
 
 PKG_NAME:=apache
-PKG_VERSION:=2.2.31
+PKG_VERSION:=2.4.20
 PKG_RELEASE:=1
 PKG_SOURCE_NAME:=httpd
 PKG_MAINTAINER:=Thomas Heil <heil@terminal-consulting.de>
 PKG_LICENSE:=Apache License
 
 PKG_SOURCE:=$(PKG_SOURCE_NAME)-$(PKG_VERSION).tar.bz2
 PKG_SOURCE_URL:=@APACHE/httpd/
-PKG_MD5SUM:=6c10e15835ab214464228a9beb7afba8
+PKG_MD5SUM:=3ff4f00ee051cf5d5304b47b1bc418b8
 
 PKG_BUILD_DIR:=$(BUILD_DIR)/$(PKG_SOURCE_NAME)-$(PKG_VERSION)
 
 PKG_INSTALL:=1
```

**005-httpd_conf.patch**
```
@@ -1,54 +1,54 @@
 --- a/docs/conf/httpd.conf.in
 +++ b/docs/conf/httpd.conf.in
-@@ -51,7 +51,6 @@ Listen @@Port@@
+@@ -63,7 +63,6 @@ Listen @@Port@@
  # Example:
  # LoadModule foo_module modules/mod_foo.so
  #
 -@@LoadModule@@
  
- <IfModule !mpm_netware_module>
- <IfModule !mpm_winnt_module>
-@@ -63,8 +62,8 @@ Listen @@Port@@
+ <IfModule unixd_module>
+ #
+@@ -74,8 +73,8 @@ Listen @@Port@@
  # It is usually good practice to create a dedicated user and group for
  # running httpd, as with most system services.
  #
 -User daemon
 -Group daemon
 +User nobody
 +Group nogroup
  
  </IfModule>
- </IfModule>
-@@ -191,7 +190,7 @@ ErrorLog "@rel_logfiledir@/error_log"
+
+@@ -188,7 +187,7 @@ ErrorLog "@rel_logfiledir@/error_log"
  # Possible values include: debug, info, notice, warn, error, crit,
  # alert, emerg.
  #
 -LogLevel warn
 +LogLevel debug
  
  <IfModule log_config_module>
      #
-@@ -336,7 +335,7 @@ DefaultType text/plain
+@@ -321,7 +320,7 @@ LogLevel warn
[... 34 lines cut ...]
         if (TEST_CHAR(c, T_ESCAPE_URLENCODED)) {
                                                ^
util.c:1926:48: note: each undeclared identifier is reported only once for each function it appears in
util.c: In function 'ap_append_pid':
util.c:2397:28: warning: format '%ld' expects argument of type 'long int', but argument 5 has type 'pid_t {aka int}' [-Wformat=]
     return apr_psprintf(p, "%s%s%" APR_PID_T_FMT, string,
                            ^
/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-x86_64_musl-1.1.14/httpd-2.4.20/build/rules.mk:206: recipe for target 'util.lo' failed
make[5]: *** [util.lo] Error 1
make[5]: Leaving directory '/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-x86_64_musl-1.1.14/httpd-2.4.20/server'
/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-x86_64_musl-1.1.14/httpd-2.4.20/build/rules.mk:75: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-x86_64_musl-1.1.14/httpd-2.4.20/server'
/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-x86_64_musl-1.1.14/httpd-2.4.20/build/rules.mk:75: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-x86_64_musl-1.1.14/httpd-2.4.20'
Makefile:164: recipe for target '/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-x86_64_musl-1.1.14/httpd-2.4.20/.built' failed
make[2]: *** [/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-x86_64_musl-1.1.14/httpd-2.4.20/.built] Error 2
make[2]: Leaving directory '/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/feeds/packages/net/apache'
package/Makefile:187: recipe for target 'package/feeds/packages/apache/compile' failed
make[1]: *** [package/feeds/packages/apache/compile] Error 2
make[1]: Leaving directory '/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64'
/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/include/toplevel.mk:186: recipe for target 'package/apache/compile' failed
make: *** [package/apache/compile] Error 2
```

Xotic750, 2016-06-03:
I also tried v2.4.18 and had the same error as above.

Vicinzu, 2016-10-18:
Here a possible solution. I was able to compile it, but had no time to test it.

The patch 002-test_char_h.patch has to be updated: 
[002-test_char_h.patch.zip](https://github.com/openwrt/packages/files/537273/002-test_char_h.patch.zip)

And the definition of apache/install should be adapted to 2.4:
[Makefile.zip](https://github.com/openwrt/packages/files/537274/Makefile.zip)

mhei, 2019-03-26:
This issue is mixing different things:
1) a potentially unsafe configuration example in the Wiki
2) a CVE issue of php
3) upgrading apache package to a newer version

Meanwhile 2. and 3. are actually solved:
* apache is available in version 2.4.37 and builds for all platform without errors on buildbots
* PHP was upgraded so that this CVE was fixed

Since the referenced Wiki is page is read-only and only available for archive purposes, we cannot do much regarding 1. I did not check whether a/the new Wiki has a better configuration example, but I'd close this issue for house-keeping reasons.

closed 2019-03-26 by mhei