openwrt/packages #2797: uhttpd: Certificate chain not being walked, FireFox and SSL Labs fail.

Xotic750, 2016-06-01:
I wanted to use uhttp as a web facing service, it was already installed with SSL enabled for luci, I soon found problems with certificates (signed by [Let's Encrypt](https://letsencrypt.org/)) in that it was not honouring a full walk of the certificate chain, whatever I tried and I could only manage a B grade result with [SSL Labs](https://www.ssllabs.com/ssltest/).  Firefox also complained that the site was unsafe as the certificate chain was not being walked. I have use the same certificate on Apache and Nginx and some others, and they have worked just fine on those.

I see than several years people were having the same problem with uhttpd, an example on the forums: https://forum.openwrt.org/viewtopic.php?id=29120 there was also ticket raised an the problem closed as fixed: https://dev.openwrt.org/ticket/12457

jow-, 2016-06-01:
Cannot confirm that, uhttpd with ustream-polarssl backend delivers the entire cert chain. Did you concat your domain cert and the intermediate X3 cert into a single pem file?

Xotic750, 2016-06-01:
Yes, I also tried the fullroot cert (all supplied as pem) and many other things like converting to .key and .crt format. Some attempts, the certificate was not accepted so they were definitely incorrect. But those attempts that were acceptable and included the full chain (to the best of my knowledge as they worked with other apps) worked fine with Chromium (as in it did not complain and said that the site was secure), but both Firefox and SSL Labs said that the full chain was unavailable and therefore the site was not fully secure. I am going to try another free SSL certificate provider, as soon as I have a little time, jut be sure that it is not a problem with those provided by Let's Encrypt (creating one with [StartSSL](https://www.startssl.com/)). The certs are 4096 bit, and I have also tried tried with a 4096 DH cert concat-ed. Maybe the problem is mine but it has not been so difficult to apply them, at least it has not been with any other app. :/

Xotic750, 2016-06-01:
I have now tried another signed certificate. The results.

https://www.ssllabs.com/ssltest/analyze.html?d=baldr.suka.se&latest

I have left uhttpd facing the internet for a little while, so you can connect, if that helps.

https://baldr.suka.se

jow-, 2016-06-01:
It would be more interesting to know how you configured it. Also what ustream backend is in use?

Xotic750, 2016-06-01:
> ibustream-polarssl 2016-02-21-6ba14990e6..d75

Isn't that what you mean?

After concat of domain and intermediate pem files, I converted to DER using openssl.
`openssl x509 -outform der -in 1_baldr.suka.se_bundle.crt -out certificate.der`

I also converted the private key with openssl
`openssl rsa -inform pem -outform der -in unenc.key -out private.der`

I then uploaded these through the luci interface and restarted the service.

**Config**
```
config uhttpd 'main'
        option redirect_https '1'
        option home '/www'
        option max_requests '3'
        option max_connections '100'
        option cgi_prefix '/cgi-bin'
        option script_timeout '60'
        option network_timeout '30'
        option http_keepalive '20'
        option tcp_keepalive '1'
        option ubus_prefix '/ubus'
        list listen_http '0.0.0.0:8080'
        list listen_https '0.0.0.0:8443'
        option rfc1918_filter '1'
        option cert '/etc/luci-uploads/cbid.uhttpd.main.cert'
        option key '/etc/luci-uploads/cbid.uhttpd.main.key'

config cert 'px5g'
        option days '730'
        option bits '2048'
        option country 'SE'
        option state 'J..rf..lla'
        option location 'Kallh..ll'
        option commonname '192.168.1.1'
```

jow-, 2016-06-01:
I mean just "cat mydomain.pem intermediate.pem > public.pem" then use `public.pem` in uhttpd. No conversion to DER, no other extra steps.

Xotic750, 2016-06-01:
I only converted to DER because that is what it states in the luci GUI and the documents.

https://wiki.openwrt.org/doc/howto/certificates.overview

I have now uploaded the PEM versions. Chromium sees it the same as the screenshot above, but now SSL Labs sees the whole thing too and so does Firefox.

https://www.ssllabs.com/ssltest/analyze.html?d=baldr.suka.se&latest

ChrisEt, 2020-07-09:
I can confirm that uhttpd also accepts PEM files. That's the way for supplying a certificate chain.

(Tested on OpenWrt 19.07.3)