openwrt/packages #2803: https://code.google.com/p/squeezelite/

schmurtzm, 2016-06-02:
Hi, the package source is not good anymore in the [Makefile](https://github.com/openwrt/packages/blob/010465f2faef8aeaf67f8b872d1d5b9d4baa13c2/sound/squeezelite/Makefile) of squeezelite.

`https://code.google.com/p/squeezelite/`

Should be probably replaced by :

`https://github.com/ralph-irving/squeezelite`

Thanks !

schmurtzm, 2016-06-02:
Or may be replaced by `https://github.com/ralph-irving/triode-squeezelite`
Which is "Mirror of googlecode squeezelite source repository https://code.google.com/p/squeezelite"

schmurtzm, 2016-06-02:
By the way it can be interesting to integrate [SqueezeWRT](https://github.com/squeezewrt/squeezewrt).
https://github.com/squeezewrt/squeezewrt
SqueezeWRT regroups squeezelite + a luci interface to control squeezelite + LIRC for remote control. The luci interface is very usefull :)
Description and details here :
https://forum.openwrt.org/viewtopic.php?id=56627

thess, 2016-06-05:
Squeezelite updated to ralph-irving fork v1.8.4-743 - see https://github.com/openwrt/packages/commit/b4fb194299b168754ee908a95a6c11a5e3c8f18b

The squeezewrt/luci interface needs to be repackaged for the luci repo and its standards.

closed 2016-06-05 by thess