openwrt/packages #2806: python/python3 pip & setuptools use the same PKG_BUILD_DIR

zyxmon, 2016-06-03:
pip & setuptools have the same version number for python2.7 and python3 and have the same PKG_BUILD_DIR's - so parallel builds may fail because of it.

zyxmon, 2016-06-04:
This is the simplest solution - https://github.com/Entware-ng/entware-packages/commit/8bd656d962320f32eceab709f474ba6a9b51383f

hnyman, 2016-06-04:
@commodo 
any views on this?

commodo, 2016-06-04:
yep ; it's on my todo list for today ; was busy during the week

commodo, 2016-06-04:
ah wait ; saw the commits now ; that's a bit of a smaller change than i had in mind
@zyxmon did you try it and it worked ?
i'm not too savy with too many tar commands

I'd also extend the solution to python-pip & python-setuptools ; it would make them easier to find them in the build_dir

zyxmon, 2016-06-04:
@commodo - yes it works in Entware. It should also work in Openwrt.

commodo, 2016-06-04:
ok; cool :)
feel free to spin up PRs then

commodo, 2016-06-04:
@zyxmon  are you also doing the changes for python-pip & python-setuptools ?
this is so that we don't collide ;

zyxmon, 2016-06-04:
https://github.com/openwrt/packages/pull/2812
No, I do not change python-pip & python-setuptools

zyxmon, 2016-06-04:
https://github.com/openwrt/packages/pull/2813

hnyman, 2016-06-09:
This can be closed, right?

commodo, 2016-06-09:
yes

closed 2016-06-09 by hnyman