openwrt/packages #2818: How to deal with not-yet-answered PRs?

bk138, 2016-06-05:
This is more like a meta-issue, sry for making that much noise but I couldn't find any docs abtout this. 

I have a PR https://github.com/openwrt/packages/pull/2626 that's been lingering for over a month now. Question: What is the preferred way to ask about PR status?

jow-, 2016-06-05:
Hi @bk138 - no it is fine to make fuzz about it and I am sorry that nobody took care of it until now :(
The lack of any feedback indicates that your PR is formally fine - I guess not too many people deem a standalone `gzip` to be important so noone bothered to comment on it.

jow-, 2016-06-05:
I think it boils down to the fact that we need more repository wide maintainers.
@openwrt/owners - any suggestions?

jow-, 2016-06-05:
Pinging @openwrt/package-maintainers too.

bk138, 2016-06-05:
Pinging those seems like a way to go, thanks

closed 2016-06-05 by bk138