openwrt/packages #2835: selecting transmission-daemon-openssl does not allow transmission-web to be selected

neheb, 2016-06-11:
AFAIK the -polarssl variant is the same.

mstorchak, 2016-06-11:
-polarssl works, in contrast to -openssl

mstorchak, 2016-06-11:
This hack worked for me. DEPENDS:= seems to not work correctly, and || (or) doesn't get parsed, that's why PACKAGE_ prefix is required.
```
diff --git a/net/transmission/Makefile b/net/transmission/Makefile
index 23279ad..24ba84d 100644
--- a/net/transmission/Makefile
+++ b/net/transmission/Makefile
@@ -36,7 +36,6 @@ define Package/transmission-daemon/Default
   $(call Package/transmission/template)
   DEPENDS:=+libcurl +libpthread +libevent2 +librt
   USERID:=transmission=224:transmission=224
-  PROVIDES:=transmission-daemon
 endef

 define Package/transmission-daemon-openssl
@@ -94,7 +93,7 @@ endef
 define Package/transmission-web
   $(call Package/transmission/template)
   TITLE+= (webinterface)
-  DEPENDS:=transmission-daemon
+  DEPENDS:=transmission-daemon-openssl||PACKAGE_transmission-daemon-polarssl
 endef
```

obsy, 2016-06-11:
@mstorchak: good idea, please make PR.

PR #2837 (2016-06-13): transmission: fix transmission-web dependencies
Fixes #2835
commit e08324cb (2016-06-13): transmission: fix transmission-web dependencies

Fixes #2835
closed 2016-06-13 by champtar