openwrt/packages #2862: krb5: libtirpc will not build due to removed libs in Build/InstallDev

Xotic750, 2016-06-15:
I don't know the reason as to why certain libs are being removed.
```
define Build/InstallDev
	$(INSTALL_DIR) $(1)/usr/include
	$(CP) $(PKG_INSTALL_DIR)/usr/include \
		$(1)/usr/include/krb5
	$(INSTALL_DIR) $(1)/usr
	$(CP) $(PKG_INSTALL_DIR)/usr/lib \
		$(1)/usr
	rm -f $(1)/usr/lib/libcom_err*
endef
```
But to compile libtirpc against krb5, these libs are required and this change fixes it.
```
define Build/InstallDev
	$(INSTALL_DIR) $(1)/usr/include
	$(CP) $(PKG_INSTALL_DIR)/usr/include \
		$(1)/usr/include/krb5
	$(INSTALL_DIR) $(1)/usr
	$(CP) $(PKG_INSTALL_DIR)/usr/lib \
		$(1)/usr
	#rm -f $(1)/usr/lib/libcom_err*
endef
```
I would be happy to submit a PR if this is not an issue for some reason?

jow-, 2016-06-15:
I think because `libcom_err` is provided by another package already.

jow-, 2016-06-15:
@Xotic750 - ah yes, package/utils/e2fsprogs implements a standalone `libcom_err` package so try to add this to your libtirpc:

`PKG_BUILD_DEPENDS:=e2fsprogs`

Xotic750, 2016-06-15:
@jow- Great, I will give that a try.

Xotic750, 2016-06-15:
That seems to have solved it. :)

Xotic750, 2016-06-15:
Even though libtirpc compiled without complaint, now compiling rpcbind I get the following error
```
ccache_cc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-x86_64_musl-1.1.14/rpcbind-0.2.3:rpcbind-0.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -L/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-x86_64_musl-1.1.14/usr/lib -L/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-x86_64_musl-1.1.14/lib -L/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/usr/lib -L/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/lib -znow -zrelro  -o rpcbind src/check_bound.o src/pmap_svc.o src/rpcb_stat.o src/rpcb_svc.o src/rpcb_svc_4.o src/rpcb_svc_com.o src/rpcbind.o src/security.o src/util.o src/warmstart.o src/xlog.o -L/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-x86_64_musl-1.1.14/usr/lib -ltirpc  
/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/bin/../lib/gcc/x86_64-openwrt-linux-musl/5.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: warning: libcom_err.so.3, needed by /home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-x86_64_musl-1.1.14/usr/lib/libgssapi_krb5.so.2, not found (try using -rpath or -rpath-link)
/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-x86_64_musl-1.1.14/usr/lib/libgssapi_krb5.so.2: undefined reference to `remove_error_table@com_err_3_MIT'
/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-x86_64_musl-1.1.14/usr/lib/libgssapi_krb5.so.2: undefined reference to `add_error_table@com_err_3_MIT'
/home/graham/Downloads/LEDE-SDK-x86-64_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-x86_64_musl-1.1.14/usr/lib/libgssapi_krb5.so.2: undefined reference to `error_message@com_err_3_MIT'
collect2: error: ld returned 1 exit status
```
krb5 creates `libcom_err.so.3` but e2fsprogs creates `libcom_err.so.0`. Is there a simple solution?

Xotic750, 2016-06-15:
I bumped e2fsprogs from v1.42.13 to v1.43.1 (which I could send a PR for) but it made no difference as it still builds `libcom_err.so.0`

Xotic750, 2016-06-15:
If I change krb5 Makefile
```
	#rm -f $(1)/usr/lib/libcom_err*
	rm -f $(1)/usr/lib/libcom_err.so
```
then I am able to build, but there is no longer a `libcom_err.so` in the `staging_dir` but there are both `libcom_err.so.0` and `libcom_err.so.3` and now everything builds without error. However, I have no idea what effect this has on any other packages or if this is the wrong way to deal with it or anything. Tried to search a little on the subject but was unable to find any useful info.

Xotic750, 2016-06-16:
Ok, I did a little more investigation and made the following changes in the the krb5 Makefile.

Added `e2fsprogs` as a dependency
```
DEPENDS:=+libncurses +e2fsprogs
```
added the `e3fsprogs` includes for `com_err.h`
```
TARGET_CPPFLAGS += -I$(STAGING_DIR)/usr/include/et
```
added `--with-system-et` to `CONFIGURE_ARGS`

> --with-system-et
> Use an installed version of the error-table (et) support software, the compile_et program, the com_err.h header file and the com_err library. If these are not in the default locations, you may wish to specify CPPFLAGS=-I/some/dir and LDFLAGS=-L/some/other/dir options at configuration time as well.
> 
> If this option is not given, a version supplied with the Kerberos sources will be built and installed along with the rest of the Kerberos tree, for Kerberos applications to link against.

Everything compiles without error. If this looks correct then I will submit a PR for krb5?

closed 2016-06-16 by Xotic750

Xotic750, 2016-06-25:
Curiosity: Should `e2fsprogs` have a `PROVIDES` entry, i.e.
```
PROVIDES:=libcom_err
```
or something similar?

I didn't find anything about it in https://wiki.openwrt.org/doc/devel/packages