openwrt/packages #2871: librpc: Can this be used instead of libtirpc?

Xotic750, 2016-06-17:
`portmap` does not appear to support IPv6, `rpcbind` appears to be it's replacement and does support this. To build `rpcbind` it depends on `libtirpc`, but all `rpc` dependant packages in OpenWRT/LEDE appear to use `librpc` (a fork from `uClibc`). Would using `librpc` for `rpcbind` (if the API is compatible) continue to allow IPv6 support and does `librpc` allow support of `krb5` like with `libtirpc`? I tried to find information but have not had any success.

Andy2244, 2018-08-08:
@Xotic750 I'm working on this currently, trying to remove librpc completely and replacing it with libtirpc. Just finished creating the rpcbind package.

Andy2244, 2018-08-13:
@Xotic750 @tripolar 

This can be closed as well.

closed 2018-08-16 by jow-