openwrt/packages #2872: Use updated version of Reaver

JarrettR, 2016-06-17:
Currently, the included version of Reaver is 1.4, which is 4 years old.

The latest community-updated version of Reaver is 1.5.2, located here:

https://github.com/t6x/reaver-wps-fork-t6x

It includes the Pixie dust attack and other improvements, and has widely been adopted as the de facto standard Reaver version. See:
https://www.kali.org/penetration-testing/pixiewps-reaver-aircrack-ng-updates/

yousong, 2016-06-21:
I have build-tested both of them [1,2] since last time @hnyman reported of reaver's broken download link (https://github.com/openwrt/packages/issues/2749), but yet cannot find time to run-test them.  Will do in the near future and make it work in OpenWrt.

Thanks for bringing this up :)

\[1] https://github.com/yousong/build-scripts/blob/master/build-reaver-wps-fork-t6x.sh
\[2] https://github.com/yousong/build-scripts/blob/master/build-pixiewps.sh

yousong, 2016-08-24:
An update on this one: transition to the fork by t6x may not happen in the near future unless someone who knows the code steps up and provides a patch.

The reason is mainly that I have no time and no need to do the transition at the moment.  An issue was opened quite a while ago (https://github.com/t6x/reaver-wps-fork-t6x/issues/66) in hope that patches in this repo can be adapted by current maintainer of the reaver fork, but it received no response at all to this moment.  Maintaining patches for actively maintained code I have little knowledge of is error-prone.

So I am going to close this one

closed 2016-08-24 by yousong

wiire-a, 2017-12-02:
@yousong 
Have you considered updating? Reaver and pixiewps have improved significantly overtime.
Especially Reaver which has a lot of fixes over the discontinued 1.4 version. See https://github.com/t6x/reaver-wps-fork-t6x/releases.

There aren't anymore endianness issues and the only dependency is *libpcap*.

Pixiewps on the other hand only requires *libpthread* (openssl is optional). A simple [Makefile](https://github.com/wiire-a/pixiewps/blob/master/Makefile) is already in the repository.

yousong, 2017-12-04:
Hi, @wiire-a , thanks for the heads-up.  It may take a while before I can find time to update the package to use the fork and incorporate pixiewps.  If you already have a plan in mind, feel free to open a pull request to do this and maybe also take the maintainership ;)

wiire-a, 2017-12-04:
@yousong 
Thank you for the reply. I don't have a build environment already set up and I'm pretty busy at the moment. My "plan" was actually to take a break from the development. Maybe later this month I can try to do some testing.

I don't know if taking the maintainership is a good idea as I don't frequently use OpenWrt. In any case if there's any problem with Reaver or pixiewps let us know.

yousong, 2017-12-05:
I just pushed changes to switch to the reaver-t6x fork and created a package for pixiewps.  I only did simple run-tests with them and no functional test was done yet.

rofl0r, 2017-12-24:
@yousong i added a configure switch `--enable-savetocurrent` which replaces `0002-Use-the-current-directory-for-storing-and-loading-se.patch` - i think you want to use that

rofl0r, 2017-12-24:
oh and btw: rather than using the funky github "tarball-from-tag" tar.gz downloads, you should use our release tarballs (tar.xz) with published checksum on that release page.

yousong, 2017-12-25:
@rofl0r , `--enable-savetocurrent` was just enabled in commit https://github.com/openwrt/packages/commit/9853f20b0fb576f54a9ff4bab6f517de826394fb .

Both reaver-wps-t6x-fork and pixiewps are already using maintainer uploaded tar.xz file but with sha256sum instead of sha512sum in the release page.

I am locking this thread now.  Feel free to open a new one if there are further issues ;)