openwrt/packages #2880: last aria2 1.24.0 - bad md5sum

build000, 2016-06-20:
`make[3]: Entering directory '/home/slawekj/Pulpit/Projects/LEDE-WNDR4300/feeds/packages/net/aria2'
mkdir -p /home/slawekj/Pulpit/Projects/LEDE-WNDR4300/dl
SHELL= flock /home/slawekj/Pulpit/Projects/LEDE-WNDR4300/tmp/.aria2-1.24.0.tar.bz2.flock -c '	/home/slawekj/Pulpit/Projects/LEDE-WNDR4300/scripts/download.pl "/home/slawekj/Pulpit/Projects/LEDE-WNDR4300/dl" "aria2-1.24.0.tar.bz2" "56c10ae766c7344393f7b0d9f5dadc5b" "" "https://github.com/aria2/aria2/releases/download/release-1.24.0/"'
--2016-06-20 22:21:19--  https://github.com/aria2/aria2/releases/download/release-1.24.0/aria2-1.24.0.tar.bz2
Resolving github.com (github.com)... 192.30.252.122
Connecting to github.com (github.com)|192.30.252.122|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-cloud.s3.amazonaws.com/releases/1116542/cbff8412-3346-11e6-99b3-5ceb48158987.bz2?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160620%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160620T222121Z&X-Amz-Expires=300&X-Amz-Signature=7b60fb1396a47bf8337c471b42f6b046ddceefe5e78e5a93d8cb044988393af4&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Daria2-1.24.0.tar.bz2&response-content-type=application%2Foctet-stream [following]
--2016-06-20 22:21:21--  https://github-cloud.s3.amazonaws.com/releases/1116542/cbff8412-3346-11e6-99b3-5ceb48158987.bz2?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160620%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160620T222121Z&X-Amz-Expires=300&X-Amz-Signature=7b60fb1396a47bf8337c471b42f6b046ddceefe5e78e5a93d8cb044988393af4&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Daria2-1.24.0.tar.bz2&response-content-type=application%2Foctet-stream
Resolving github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)... 54.231.32.91
Connecting to github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)|54.231.32.91|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2264145 (2.2M) [application/octet-stream]
Saving to: 'STDOUT'

-                                  100%[================================================================>]   2.16M   298KB/s    in 14s     

2016-06-20 22:21:38 (155 KB/s) - written to stdout [2264145/2264145]

MD5 sum of the downloaded file does not match (file: d8d75cd0ee01c3e50e57214ff552a961, requested: 56c10ae766c7344393f7b0d9f5dadc5b) - deleting download.
--2016-06-20 22:21:38--  http://sources.lede-project.org/aria2-1.24.0.tar.bz2
Resolving sources.lede-project.org (sources.lede-project.org)... 148.251.78.235, 2a01:4f8:202:43ea::2
Connecting to sources.lede-project.org (sources.lede-project.org)|148.251.78.235|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-06-20 22:21:40 ERROR 404: Not Found.

Download failed.
--2016-06-20 22:21:40--  http://mirror2.openwrt.org/sources/aria2-1.24.0.tar.bz2
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-06-20 22:21:41 ERROR 404: Not Found.

Download failed.
--2016-06-20 22:21:41--  http://downloads.openwrt.org/sources/aria2-1.24.0.tar.bz2
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-06-20 22:21:43 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
Makefile:81: recipe for target '/home/slawekj/Pulpit/Projects/LEDE-WNDR4300/dl/aria2-1.24.0.tar.bz2' failed
make[3]: *** [/home/slawekj/Pulpit/Projects/LEDE-WNDR4300/dl/aria2-1.24.0.tar.bz2] Error 2
make[3]: Leaving directory '/home/slawekj/Pulpit/Projects/LEDE-WNDR4300/feeds/packages/net/aria2'
package/Makefile:187: recipe for target 'package/feeds/packages/aria2/compile' failed
make[2]: *** [package/feeds/packages/aria2/compile] Error 2
make[2]: Leaving directory '/home/slawekj/Pulpit/Projects/LEDE-WNDR4300'
package/Makefile:184: recipe for target '/home/slawekj/Pulpit/Projects/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.14/stamp/.package_compile' failed
make[1]: *** [/home/slawekj/Pulpit/Projects/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.14/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/slawekj/Pulpit/Projects/LEDE-WNDR4300'
/home/slawekj/Pulpit/Projects/LEDE-WNDR4300/include/toplevel.mk:193: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
`

build000, 2016-06-20:
Please use this link https://github.com/aria2/aria2/releases/download/release-1.24.0/aria2-1.24.0.tar.bz2 - prepare,compile and build properly.

kuoruan, 2016-06-21:
Thank you for notice me, It's my fault. I just commited a new request pull.

PR #2881 (2016-06-21): aria2: fix md5sum
- fix #2880
closed 2016-06-21 by yousong