openwrt/packages #2884: Why the torrent files can't be download with aria2 on  openwrt?

luofeiyu, 2016-06-22:
The test torrent file located on:
https://dl.dropboxusercontent.com/u/93374194/se.torrent

For the soft --aria2.

It can be executed on my desktop,no problem.

    aria2c /tmp/se.torrent  -d /tmp -o se.iso

The same command failed  on my openwrt.

   aria2c /tmp/se.torrent  -d /tmp -o se.iso
    Exception caught
    Exception: [download_helper.cc:411] errorCode=1 Unrecognized URI or unsupported protocol: /tmp/se.torrent

Please try it with the test torrent on `https://dl.dropboxusercontent.com/u/93374194/se.torrent`.

FriedZombie, 2016-06-22:
Seems like there is no torrent support is compiled in aria2 in the openwrt package

hnyman, 2016-06-22:
Bittorrent support in aria2 is off by default.
If you need that feature, you can change the option in your build config and re-compile aria2 by yourself.
https://github.com/openwrt/packages/blob/master/net/aria2/Config.in#L22

closed 2016-06-22 by hnyman