openwrt/packages #2889: miniupnpc lease time broken

axet, 2016-06-23:
Hello!

I setup home openwrt router with miniupnpd, and debuging how external port assigned. Not sure who to blame yet, I have to check source, but seems like miniupnp does not update lease time when I send repeated requests 'AddPortMapping'

To get port I have to provide port information and lease time. For example when I ask to keep port for 5 minutes, I keep in mind I will request port again after 1 minute to keep port assigned to me without interruptions.

Using cat /tmp/upnp.leases I can check how long leases stands and find out lease time not been updated after my repeaded calls of 'AddPortMapping' requests.

I'm using https://github.com/syncthing/syncthing/blob/master/lib/upnp library, which should be ok.

@sbyx

axet, 2016-06-24:
This bug was fixed here

https://github.com/miniupnp/miniupnp/commit/430dcc7b22d593295150210832d29a3912a509b2

axet, 2016-06-27:
@fingon

kdarbyshirebryant, 2016-06-27:
This https://patchwork.ozlabs.org/patch/627484/ has been busy bit-rotting since 29th May.

axet, 2016-06-27:
@kdarbyshirebryant  It seems broken. Are't maintainers have to fix well know bugs? What is happening?

fingon, 2016-06-27:
This is purely volunteer work so we do not _have_ to do anything_. Regardless, thanks for the patch, I merged yours to for-15.05 (and committed similar fix to master as well). I do not typically merge anything without testing it and I have not had the time to look at this stuff in awhile.

closed 2016-06-28 by axet

axet, 2016-06-28:
@fingon don't take me wrong, openwrt takes responsibility for all users using it. This includes keeping software free of known bugs and stability. And this responsibility distributed between maintainers.