openwrt/packages #2890: procps-ng fails to build

ffainelli, 2016-06-23:
Hi @notnyt ,

Looks like your update of procps-ng in commit e08d088cd5ed82954cb3e7bace35e364c3fa9f44 breaks the build for me:

remote: Counting objects: 9806, done.
remote: Compressing objects: 100% (2603/2603), done.
remote: Total 9806 (delta 7271), reused 9544 (delta 7080)
Receiving objects: 100% (9806/9806), 3.76 MiB | 1.42 MiB/s, done.
Resolving deltas: 100% (7271/7271), done.
Checking connectivity... done.
Note: checking out '8a198e3eafbde0b627e9992b41d26e6762e4f8c7'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 8a198e3eafbd... ps: adapt to revised approach for --with-systemd build
Packing checkout...
. /home/fainelli/dev/openwrt/trunk/include/shell.sh; gzip -dc /home/fainelli/dev/openwrt/trunk/dl/procps-ng-3.3.11-8a198e3eafbde0b627e9992b41d26e6762e4f8c7.tar.gz | tar -C /home/fainelli/dev/openwrt/trunk/build_dir/target-arm_cortex-a9_glibc-2.22_eabi/procps-ng-3.3.11/.. -xf - 
touch /home/fainelli/dev/openwrt/trunk/build_dir/target-arm_cortex-a9_glibc-2.22_eabi/procps-ng-3.3.11/.prepared_618b5ae054295e176607dbef74bd0baf
rm -f /home/fainelli/dev/openwrt/trunk/staging_dir/target-arm_cortex-a9_glibc-2.22_eabi/stamp/.procps-ng_installed
(cd /home/fainelli/dev/openwrt/trunk/build_dir/target-arm_cortex-a9_glibc-2.22_eabi/procps-ng-3.3.11; echo "3.3.11" > "/home/fainelli/dev/openwrt/trunk/build_dir/target-arm_cortex-a9_glibc-2.22_eabi/procps-ng-3.3.11/.tarball-version"; ./autogen.sh );
You must have autopoint installed to generate procps-ng build system.
The autopoint command is part of the GNU gettext package.
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /home/fainelli/dev/openwrt/trunk/staging_dir/host/bin/automake line 3938.
Makefile:96: recipe for target '/home/fainelli/dev/openwrt/trunk/build_dir/target-arm_cortex-a9_glibc-2.22_eabi/procps-ng-3.3.11/.configured_yyyn' failed
make[3]: *** [/home/fainelli/dev/openwrt/trunk/build_dir/target-arm_cortex-a9_glibc-2.22_eabi/procps-ng-3.3.11/.configured_yyyn] Error 1
make[3]: Leaving directory '/home/fainelli/dev/github/openwrt/packages/utils/procps-ng'
package/Makefile:196: recipe for target 'package/feeds/packages/procps-ng/compile' failed
make[2]: *** [package/feeds/packages/procps-ng/compile] Error 2
make[2]: Leaving directory '/home/fainelli/dev/openwrt/trunk'
package/Makefile:193: recipe for target '/home/fainelli/dev/openwrt/trunk/staging_dir/target-arm_cortex-a9_glibc-2.22_eabi/stamp/.package_compile' failed
make[1]: *** [/home/fainelli/dev/openwrt/trunk/staging_dir/target-arm_cortex-a9_glibc-2.22_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/fainelli/dev/openwrt/trunk'

Depending on autopoint to build this package should mean building it as part of the host tool build process.

Thanks

jow-, 2016-06-23:
This is likely because the makefile invokes autogen.sh, try using the auroreconf fixup instead

closed 2016-07-02 by champtar

stsydow, 2016-07-07:
Works for ar71xx but is broken on malta targets:
```
autoreconf: Entering directory `.'
autoreconf: running: true --force
autoreconf: running: /home/st/dev/lede/staging_dir/host/bin/aclocal -I /home/st/dev/lede/staging_dir/target-mips_24kc_musl-1.1.14/host/share/aclocal -I /home/st/dev/lede/staging_dir/target-mips_24kc_musl-1.1.14/usr/share/aclocal -I m4 -I . --force -I m4
configure.ac:72: warning: macro 'AM_GNU_GETTEXT_VERSION' not found in library
configure.ac:73: warning: macro 'AM_GNU_GETTEXT' not found in library
autoreconf: configure.ac: tracing
autoreconf: configure.ac: AM_GNU_GETTEXT_VERSION is used, but not AM_GNU_GETTEXT
autoreconf: running: /home/st/dev/lede/staging_dir/host/bin/libtoolize --force
OpenWrt-libtoolize: putting auxiliary files in `.'.
OpenWrt-libtoolize: linking file `./ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
OpenWrt-libtoolize: linking file `m4/libtool.m4'
OpenWrt-libtoolize: linking file `m4/ltoptions.m4'
OpenWrt-libtoolize: linking file `m4/ltsugar.m4'
OpenWrt-libtoolize: linking file `m4/ltversion.m4'
OpenWrt-libtoolize: linking file `m4/lt~obsolete.m4'
configure.ac:72: warning: macro 'AM_GNU_GETTEXT_VERSION' not found in library
configure.ac:73: warning: macro 'AM_GNU_GETTEXT' not found in library
autoreconf: running: /home/st/dev/lede/staging_dir/host/bin/autoconf --include=/home/st/dev/lede/staging_dir/target-mips_24kc_musl-1.1.14/host/share/aclocal --include=/home/st/dev/lede/staging_dir/target-mips_24kc_musl-1.1.14/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/st/dev/lede/staging_dir/host/share/aclocal --force
configure.ac:72: error: possibly undefined macro: AM_GNU_GETTEXT_VERSION
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:73: error: possibly undefined macro: AM_GNU_GETTEXT
```

PR #2893 (2016-07-02): procps-ng: fix build on systems without gettext development utilities
The current procps-ng Makefile calls the shipped autogen.sh script which
introduces incorrect implicit dependencies on host utilities, leading to
the following error observed on a minimal build system:

    (cd .../procps-ng-3.3.11; echo "3.3.11" > ".../procps-ng-3.3.11/.tarball-version"; ./autogen.sh );
    You must have autopoint installed to generate procps-ng build system.
    The autopoint command is part of the GNU gettext package.
    Makefile:96: recipe for target '.../procps-ng-3.3.11/.configured_yynyyyyy' failed
   [... cut]
commit 3048ebed (2016-07-02): procps-ng: fix build on systems without gettext development utilities

The current procps-ng Makefile calls the shipped autogen.sh script which
introduces incorrect implicit dependencies on host utilities, leading to
the following error observed on a minimal build system:

    (cd .../procps-ng-3.3.11; echo "3.3.11" > ".../procps-ng-3.3.11/.tarball-version"; ./autogen.sh );
    You must have autopoint installed to generate procps-ng build system.
    The autopoint command is part of the GNU gettext package.
    Makefile:96: recipe for target '.../procps-ng-3.3.11/.configured_yynyyyyy' failed
  [... cut]