openwrt/packages #2894: pythonX-distutils: exe files packed

ryzhovau, 2016-06-25:
`python_distutils` ipk package contains:
```
/tmp # find /lib/python2.7/ -name *.exe
/lib/python2.7/distutils/command/wininst-9.0.exe
/lib/python2.7/distutils/command/wininst-9.0-amd64.exe
/lib/python2.7/distutils/command/wininst-8.0.exe
/lib/python2.7/distutils/command/wininst-7.1.exe
/lib/python2.7/distutils/command/wininst-6.0.exe
/tmp # find /opt/lib/python2.7/ -name *.exe | du -hLs
3.4M    .
```
The same with `python3_distutils`.

ryzhovau, 2016-06-26:
@commodo

commodo, 2016-06-27:
[ i don't always respond in the weekends ; especially now with the summer high temperatures ]

will take a look ; i thought those were not packaged ;
thanks for reporting

commodo, 2016-06-27:
PRs https://github.com/openwrt/packages/pull/2901  &  https://github.com/openwrt/packages/pull/2902

Thanks again

closed 2016-07-12 by champtar