openwrt/packages #2895: at: missing from 15.05 branch

Diftraku, 2016-06-25:
For some unknown reason, 'at' has been removed from the 15.05 branch.

The same package exists in 14.07 and master but not in 15.05. Unless master is 'stable' for production, pulling from master is not a solution.

yousong, 2016-08-02:
Packages feed had a transition from svn repo to git repo here.  At the time of transition, old packages (as they are called) in svn repo were not automatically imported as many of them were broken and unmaintained.   `at` package was imported to this repo here with commit https://github.com/openwrt/packages/commit/212d602c51d7f76ce6a550482e4edf6c8b161f11 , which was after the creation of `for-15.05` branch.

You can try enabling the `oldpackages` feed in `feeds.conf.default` to use `at` with `15.05`.

Diftraku, 2016-08-02:
The problem is, it no longer exists in `oldpackages` as it has been moved to the Github master, see https://git.openwrt.org/?p=packages.git;a=commit;h=87650ede3f59e5b0fa5b956ad9657018583c8f9a for the removal from `oldpackages`. Thus, `oldpackages` is not really an option.

I would rather not pull straight from master unless master can be considered stable. Due to `at` not being "ported" back to 15.05 in a timely manner, I have had to add it in a downstream repo since a build requires it.

yousong, 2016-08-02:
Okay, I understand the situation now.  15.05 branch for oldpackages is missing...  Packages imported here are to be deleted from oldpackages, and are going to be not available for old releases...

ping `at`'s current maintainer @philenotfound to see if the package in its current form is viable for backporting to 15.05 branch.

Diftraku, 2016-08-02:
I have successfully used the current version of `at` in master on a 15.05 based build. It's probably best to compile it and install it manually on a vanilla install of 15.05 to be safe.

philenotfound, 2016-08-02:
@yousong Yes, it should work fine for 15.05, as this was the release I tested it with back then.
I have noticed there have been a few updates to the source (3.1.20), I'll look into updating the package.

diizzyy, 2017-06-09:
@thess @hnyman 
Please close, 15.05 is unsupported and frozen.

closed 2017-06-09 by thess