openwrt/packages #2900: tmux fails to allow connection due to bad locale

dfc-d, 2016-06-27:
@mstorchak 

Attempting to ssh to a device which automatically starts tmux on login from ubuntu, results in the following error message:
tmux: need UTF-8 locale (LC_CTYPE) but have ANSI_X3.4-1968

It appears tmux locale handing has changed and broken.

dfc-d, 2016-06-27:
Actually this affects console and manual launch as well.

mstorchak, 2016-07-05:
If I set LC_CTYPE to "ANSI_X3.4-1968", tmux still works for me, both old 2.0 on the main system and 2.2 in a chroot.
@dfc-d, please explain how to reproduce the issue.

dfc-d, 2016-07-06:
On Mon, 2016-07-04 at 22:48 -0700, mstorchak wrote:
> If I set LC_CTYPE to "ANSI_X3.4-1968", tmux still works for me, both
> old 2.0 on the main system and 2.2 in a chroot.
> @dfc-d, please explain how to reproduce the issue.

Hmmmm...

On Ubuntu 16.04:

export LC_CTYPE='ANSI_X3.4-1968'
bash: warning: setlocale: LC_CTYPE: cannot change locale (ANSI_X3.4-
1968): No such file or directory

And I see no way to generate that locale via dpkg-reconfigure locales.

Basically the problem is that the ANSI locale is not universally
available and Ubuntu at least is almost exclusive UTF-8 with some
legacy ISO locales.

Regards,

Daniel

dfc-d, 2016-08-10:
@mstorchak Did you see my Ubuntu comment?

mstorchak, 2016-08-11:
I did, but I don't see how it helps to understand the root cause, I'm sorry.
tmux requires UTF-8 locale, "and Ubuntu at least is almost exclusive UTF-8", so everything is fine?

If not, please explain how to reproduce the issue.

Looks like the change in tmux in question is https://github.com/tmux/tmux/commit/0d6de44a37755f0e5046c04e19e4506a6d59e750, so only check method is different, but not the check subject itself.

dfc-d, 2016-08-13:
What I try to ssh to the device from Ubuntu 16.04 tmux: need UTF-8 locale (LC_CTYPE) but have ANSI_X3.4-1968 when I try to use tmux.

When I attempt to set Ubuntu local to ANSI_X.4-1968 I get the above error.

My Ubuntu locale is UTF-8.en_CA (LANG=en_CA.UTF-8), default configuration

So I ought to be in a UTF-8 locale.  Perhaps this issue is that Ubuntu doesn't include LC_CTYPE=UTF-8.en_CA in the environemnt?  I will investigate that possiblity later.

VeggieMeat, 2016-08-16:
You might try re-installing the locales package. I had a similar issue after upgrading to Debian Stretch, and that resolved it.

chinawrj, 2016-09-27:
@dfc-d Hi, Did you compile OpenWRT under ubuntu 16.04; I found the same issue when I compile the source code with ubuntu 16.04. But it was OK when I compiled with ubuntu 14.04

```
tmux: need UTF-8 locale (LC_CTYPE) but have ANSI_X3.4-1968
```

dfc-d, 2016-09-29:
I did build with 16.04 but a more recent build (newer trunk) no longer gives me this issues; not sure if the issue was solved by an update....
Do you still see the problem against latest trunk (another possibility it that the fix is in LEDE trunk rather but not OpenWrt trunk).

diizzyy, 2017-06-02:
@hnyman 
This seems resolved, close?

closed 2017-06-02 by hnyman

jerome-diver, 2017-06-19:
it is not, when i try to connect ssh with command linked tmux for attach it to a session, i have the same error.
My locales are fine in server remote computer and server side... together on archlinux.
i don't understand what is this ANSI_X3.4-1968 here ? where this is coming from ? 
when i just connect ssh all is ok... and when i call tmux on remote computer (or server side), all is ok...so why this happen ? is it a bug ?

cwbs, 2017-07-16:
I connected via ssh on archlinux and also have these errors: need UTF-8 locale (LC_CTYPE) but have ANSI_X3.4-1968

raphaunix, 2017-08-27:
This worked for me:

```
echo "en_US.UTF-8 UTF-8" >> /etc/locale.gen 
locale-gen 
```
Enjoy.

zokl, 2018-03-02:
I have the same problem but only if I compile openwrt image with glibc library. ulibc work fine. Do you know where is the problem?

princeofdream, 2021-04-15:
openwrt build with glibc does not have locale config files.
we should copy the locale files and regenerate locale info.

# 1. copy files from staging_dir/toolchain-xxx_glibc
``
cd staging_dir/toolchain-xxx_glibc
cp -rP bin/locale ${rootfs}/bin/
cp bin/localedef ${rootfs}/bin/
cp share ${rootfs}/bin/
``
# 2. make dirs for generate files
``
mkdir -p /usr/lib/locale/locale-archive/
mkdir -p /lib/locale/
``

# 3. regenerate locale config
``
localedef -v -c -i en_US -f UTF-8 en_US.UTF-8
``
then export TERM=xterm to run tmux
it works for me, but I try a minimal build and do the steps, it does not works well.

zokl, 2021-06-23:
> openwrt build with glibc does not have locale config files.
> we should copy the locale files and regenerate locale info.
> 
> # 1. copy files from staging_dir/toolchain-xxx_glibc
> `cd staging_dir/toolchain-xxx_glibc cp -rP bin/locale ${rootfs}/bin/ ; cp bin/localedef ${rootfs}/bin/ ; cp sbin/* ${rootfs}/sbin/ ; cp share ${rootfs}/bin/ ;`
> 
> # 2. make dirs for generate files
> `mkdir -p /usr/lib/locale/locale-archive/ ; mkdir -p /lib/locale/`
> 
> # 3. regenerate locale config
> `localedef -v -c -i en_US -f UTF-8 en_US.UTF-8`
> 
> then reboot;
> after that export TERM=xterm to run tmux
> it works for me

Thank you, I will try it.

zokl, 2021-06-24:
Thank you very much. Your process works. I made it in the following way and put it into my makefile:

package install section:
```
	$(INSTALL_DIR) $(1)/bin
	$(INSTALL_BIN) $(TOOLCHAIN_DIR)/bin/locale $(1)/bin/
	$(INSTALL_BIN) $(TOOLCHAIN_DIR)/bin/localedef $(1)/bin/

	$(INSTALL_DIR) $(1)/usr/share
	$(CP) $(TOOLCHAIN_DIR)/share/i18n $(1)/usr/share/
```

On the machine:
```
mkdir -p /usr/lib/locale/locale-archive/ 
mkdir -p /lib/locale/

I18NPATH="/usr/share/i18n" localedef -v -c -i en_US -f UTF-8 en_US.UTF-8

```

bobguerra, 2021-09-07:
you might not want to remotely use your Linux machine with your PowerShell with Tmux just say'n