openwrt/packages #2908: net/vpnc: include reboot disclaimer in readme

TaiSHiNet, 2016-06-29:
Hi,

You should include a 'reboot' disclaimer on the README. 
Even if you install the package you need to, else you wont get ANY output when trying to ifup

Bests

yousong, 2016-08-02:
ping vpnc maintainer @danielg4 :)

danielg4, 2016-08-02:
I keep rereading the above, but somehow, I still don't understand what you meant.

TaiSHiNet, 2016-08-02:
If you don't reboot after installing vpnc packages and try to get it working it'll just return... nothing (I mean literally nothing, even exit code 0 on commands).
So what I mean is you should add a small disclaimer, note, postit or meme saying "Please reboot after installing this module for it to be usable"

yousong, 2016-08-02:
After checking `files/vpnc.sh`, I think it's just that `netifd` needs a restart to load vpnc as a new protocol.  @TaiSHiNet , can you find time to give it a try?

danielg4, 2016-08-02:
Restarting netifd may disrupt SSH, so that's not a huge distinction in all practicality…

yousong, 2016-08-02:
Restarting netifd will cause the ssh traffic hold for a few seconds and the session should resume if everything goes okay.  I am pretty sure I have done this many times before.

TaiSHiNet, 2016-08-02:
Restarting network wont disrupt SSH, will behave as @yousong said. I'm not sure if it'll make vpnc work.
Can't test right now, router is at home.

danielg4, 2016-08-02:
The disclaimer is merged now. Oh well.

yousong, 2016-12-24:
Close this one as the real cause of issue is found and suggestions for reboots is added anyway ;)

closed 2016-12-24 by yousong