openwrt/packages #2910: stoken package is broken

dlenski, 2016-06-30:
The stoken package is broken because it installs the `stoken` shell script wrapper rather than the actual executable, which ends up in `.libs/stoken` in the build directory:

```diff
diff --git a/utils/stoken/Makefile b/utils/stoken/Makefile
index 7fcbdad..b20e18e 100644
--- a/utils/stoken/Makefile
+++ b/utils/stoken/Makefile
@@ -65,7 +65,7 @@ endef
 
 define Package/stoken/install
        $(INSTALL_DIR) $(1)/usr/bin
-       $(INSTALL_BIN) $(PKG_BUILD_DIR)/stoken $(1)/usr/bin/
+       $(INSTALL_BIN) $(PKG_BUILD_DIR)/.libs/stoken $(1)/usr/bin/
 endef
 
 define Package/libstoken/install
```

This can be seen in any of the OpenWRT 15.05 builds of this package (e.g https://downloads.openwrt.org/chaos_calmer/15.05/bcm53xx/generic/packages/packages/libstoken_0.8-1_bcm53xx.ipk, which I was trying to install). The `stoken` file in the package is a wrapper script which begins with the following (accurate!) warning:

```sh
# This wrapper script should never be moved out of the build directory.
# If it is, it will not operate correctly.
```

ffainelli, 2016-06-30:
@dlenski can you check that pull request #2911 fixes this for you?

closed 2016-06-30 by ffainelli

dlenski, 2016-06-30:
Thanks! I will test as soon as a new snapshot build shows up. (Haven't had time to get my own build environment fully set up yet :-1:.)

https://downloads.openwrt.org/snapshots/trunk/bcm53xx/generic/packages/packages/

PR #2911 (2016-06-30): stoken: Fix binary installation
Maintainer: ffainelli
Compile tested: sunxi
Run tested: sunxi

Description:
We were copying a wrapper script instead of the actual executable, fixes #2910
commit 2c80de90 (2016-06-30): stoken: Fix binary installation

We were copying a wrapper script instead of the actual executable.

Fixes #2910