openwrt/packages #2950: ddns-scripts don't exit on invalid local IP

isleros, 2016-07-10:
**Observed**: If the get_local_ip function in dynamic_dns_functions.sh returns an invalid IP address the daemon is stopped at send_update because the first argument of write_log is > 10.

 `[ -z "$__IP" ] && write_log 14 "Private or invalid or no IP '$1' given! Please check your configuration"`

**Expected**: do not exit if the IP is invalid. 
**Reason**: If a web service is used to obtain the local IP,  an invalid IP response can be a temporal network or server problem problem and the update script must not be stopped.
**Suggestion**: replace the write_log value 14 for 4

chris5560, 2016-07-13:
Looking at line 907 of dynamic_dns_fuctions.sh the function do_transfer is called if using web to detect local ip.
Inside do_transfer at line 768-794 the function run into retry loop if there are any errors detected by transfer program.
If there is no transfer error the function return whatever it gets. Now get_local_ip checks if this is a valid IP.

From my point of view it makes no sense to continue if there are problems to detect your local IP.
The local IP is the basic you need to know before thinking about changing your DNS entry in the web.

isleros, 2016-07-15:
We do not need to know the IP before because dynamic dns providers such us **dynu.net use the http request source IP** if the address is not specified as a request parameter. However **ddns-scripts requires knowing the IP in advance**. An alternative for us to solve this problem is to change the ddns-scripts logic to not require the IP in advance, this would be better because a potential point of failure (unreliable third party web sites to know your own ip.) is removed.

We lost access to two devices because the web site we use to get the IP returned 127.0.0.1 for unknown reasons, or at least that is what ddns-script logs show. The motivation to create the issues is to help to other users to prevent this problem. 

We use **http://checkip.amazonaws.com/** because it answers faster than alternatives. Is hard for us to believe that the Amazon service returned 127.0.0.1, but if that is not what happened then there is another bug somewhere.

We need to use a web site because our devices are behind NAT devices that we do not control.

chris5560, 2016-07-16:
This mean, if [IP]-placehoder is part of update url and local IP cannot be detected then fail.
otherwise ignore if local IP can be detected and send an update to the provider because local <> remote.
This do every "retry-interval" (default 10 minutes) in worst case if you cannot detect local IP.
What does your ddns provider think about this update rate ?
If you increase retry-interval you might not detect ip change before next retry.
Please do not misunderstood, I'm only think about the details of the logic before changing it. Thanks for support.

Do you know the url to check for IPv6 only on amazonaws.com because checkip.amazonaws.com only returns IPv4

isleros, 2016-07-16:
Using the presence of the [IP]-placeholder to change the update logic is a good solution.
I've asked the ddns provider what is an acceptable minimum update interval. 
If the concern is being blocked by the DDNS provider for excessively frequent updates,  then why would a DDNS provider be more restrictive than a "check my ip" web service? It looks like some "check my ip" web services already have limits. 
A feature to evaluate: allowing a list of " check my ip" web services to avoid being blocked for excessive requests, and to have alternatives if one of the services stops working or returns and invalid response.

I don't know if amazon provides a checkip service for IPv6.

I do not misunderstand, we are trying together to make an even better ddns-scripts. Thank you.

chris5560, 2016-07-21:
I think the best solution is to use a list of "checkip" web pages, because ddns-scripts can check itself if update is really necessary. It also fixes problems if a "checkip" web page goes temporary down etc.
ddns-scripts should use files like services/services_ip6 and the user can select one of them (preferred) or -custom- and if it fails ddns-scripts will error log and use automatically the next one from the file.
Additional benefit we will get a growing list of "checkip" web pages from the community.
What do you think.

isleros, 2016-07-24:
Yes, I think allowing a list of "checkip" web pages is a good improvement. Possible failures to be detected and logged by ddns-scripts include: connection timeout and reject, HTTP errors, and responses with invalid/private IP addresses.

However I prefer the option of not using a "checkip" web page at all, leaving the IP field empty for the dynamic DNS provider that support using the source IP of the HTTP connection. 

I checked with my dynamic DNS provider and they replied they don't limit updates. I prefer this solution to avoid unnecessary problems with third party web sites.