openwrt/packages #2961: luci-app-aria2 error

hostkit, 2016-07-13:
if /etc/config/aria2 not empty 
```
daemon=true
enable-rpc=true
rpc-allow-origin-all=true
rpc-listen-port=6800
rpc-listen-all=true
rpc-secret=rootopenwrt
dir=/mnt/sda5/downloads
log=/var/log/aria2.log
log-level=warn
auto-save-interval=30
event-poll=select
continue=true
input-file=/mnt/sda5/aria2.session
save-session=/mnt/sda5/aria2.session
max-concurrent-downloads=1
```

show error when open service > aria2 setting:
```
/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute cbi dispatcher target for entry '/admin/services/aria2'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:311: Unable to read UCI data: aria2
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>

```
system log:
```
Wed Jul 13 11:56:45 2016 daemon.err uhttpd[1184]: /usr/lib/lua/luci/dispatcher.lua:433: Failed to execute cbi dispatcher target for entry '/admin/services/aria2'.
Wed Jul 13 11:56:45 2016 daemon.err uhttpd[1184]: The called action terminated with an exception:
Wed Jul 13 11:56:45 2016 daemon.err uhttpd[1184]: /usr/lib/lua/luci/cbi.lua:311: Unable to read UCI data: aria2
Wed Jul 13 11:56:45 2016 daemon.err uhttpd[1184]: stack traceback:
Wed Jul 13 11:56:45 2016 daemon.err uhttpd[1184]: 	[C]: in function 'assert'
Wed Jul 13 11:56:45 2016 daemon.err uhttpd[1184]: 	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
Wed Jul 13 11:56:45 2016 daemon.err uhttpd[1184]: 	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>
```

if config empty system log
```
Wed Jul 13 12:01:44 2016 daemon.err uhttpd[1184]: /usr/lib/lua/luci/dispatcher.lua:433: Failed to execute cbi dispatcher target for entry '/admin/services/aria2'.
Wed Jul 13 12:01:44 2016 daemon.err uhttpd[1184]: The called action terminated with an exception:
Wed Jul 13 12:01:44 2016 daemon.err uhttpd[1184]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Wed Jul 13 12:01:44 2016 daemon.err uhttpd[1184]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'aria2/web_script'.
Wed Jul 13 12:01:44 2016 daemon.err uhttpd[1184]: A runtime error occured: [string "/usr/lib/lua/luci/view/aria2/web_script.htm"]:1: attempt to call global 'url' (a nil value)
Wed Jul 13 12:01:44 2016 daemon.err uhttpd[1184]: stack traceback:
Wed Jul 13 12:01:44 2016 daemon.err uhttpd[1184]: 	[C]: in function 'assert'
Wed Jul 13 12:01:44 2016 daemon.err uhttpd[1184]: 	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
Wed Jul 13 12:01:44 2016 daemon.err uhttpd[1184]: 	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>
```

kuoruan, 2016-07-13:
Before you use luci-app-aria2, you must install aria2, and /etc/config/aria2 is this file :
https://github.com/openwrt/packages/blob/master/net/aria2/files/aria2.conf

And the aria2 config you show above is in /var/etc/aria2 by default.

hostkit, 2016-07-13:
not fix:
Wed Jul 13 15:11:07 2016 daemon.err uhttpd[1183]: /usr/lib/lua/luci/dispatcher.lua:433: Failed to execute cbi dispatcher target for entry '/admin/services/aria2'.
Wed Jul 13 15:11:07 2016 daemon.err uhttpd[1183]: The called action terminated with an exception:
Wed Jul 13 15:11:07 2016 daemon.err uhttpd[1183]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Wed Jul 13 15:11:07 2016 daemon.err uhttpd[1183]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'aria2/web_script'.
Wed Jul 13 15:11:07 2016 daemon.err uhttpd[1183]: A runtime error occured: [string "/usr/lib/lua/luci/view/aria2/web_script.htm"]:1: attempt to call global 'url' (a nil value)
Wed Jul 13 15:11:07 2016 daemon.err uhttpd[1183]: stack traceback:
Wed Jul 13 15:11:07 2016 daemon.err uhttpd[1183]: 	[C]: in function 'assert'
Wed Jul 13 15:11:07 2016 daemon.err uhttpd[1183]: 	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
Wed Jul 13 15:11:07 2016 daemon.err uhttpd[1183]: 	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>

kuoruan, 2016-07-14:
I think you should run `rm -f /tmp/luci-indexcache` first, then reboot you device.

Check this file https://github.com/openwrt/luci/blob/master/applications/luci-app-aria2/luasrc/controller/aria2.lua, it contains a function status. and https://github.com/openwrt/luci/blob/master/applications/luci-app-aria2/luasrc/view/aria2/web_script.htm will call this function.

hostkit, 2016-07-14:
this is /usr/lib/lua/luci/controller/aria2.lua
```
--[[
LuCI - Lua Configuration Interface - aria2 support

Copyright 2014-2015 nanpuyue <nanpuyue@gmail.com>
Modified by kuoruan <kuoruan@gmail.com>

Licensed under the Apache License, Version 2.0 (the "License");
you may not use this file except in compliance with the License.
You may obtain a copy of the License at

	http://www.apache.org/licenses/LICENSE-2.0
]]--

module("luci.controller.aria2", package.seeall)

function index()
	if not nixio.fs.access("/etc/config/aria2") then
		return
	end

	local page = entry({"admin", "services", "aria2"}, cbi("aria2"), _("Aria2 Settings"))
	page.dependent = true

	entry({"admin", "services", "aria2", "status"}, call("status")).leaf = true

end

function status()
	local sys  = require "luci.sys"
	local ipkg = require "luci.model.ipkg"
	local http = require "luci.http"
	local uci  = require "luci.model.uci".cursor()

	local status = {
		running = (sys.call("pidof aria2c > /dev/null") == 0),
		yaaw = ipkg.installed("yaaw"),
		webui = ipkg.installed("webui-aria2")
	}

	http.prepare_content("application/json")
	http.write_json(status)
end
```
this /usr/lib/lua/luci/view/aria2/web_script.htm
```
<script type="text/javascript">//<![CDATA[
XHR.poll(5, '<%=url([[admin]], [[services]], [[aria2]], [[status]])%>', null,
	function(x, data) {
		var tb = document.getElementById('aria2_status');
		if (data && tb) {
			if (data.running) {
				var links = '<em><%:The Aria2 service is running.%></em>';
				if (data.yaaw) {
					links += '<input class="cbi-button mar-10" type="button" value="<%:Open YAAW%>" onclick="openWebUI(\'yaaw\');" />';
				}
				if (data.webui) {
					links += '<input class="cbi-button mar-10" type="button" value="<%:Open WebUI-Aria2%>" onclick="openWebUI(\'webui-aria2\');" />';
				}
				tb.innerHTML = links;
			} else {
				tb.innerHTML = '<em><%:The Aria2 service is not running.%></em>';
			}
		}
	}
);

function randomString(len) {
	len = len || 32;
	var $chars = 'abcdefghijklmnopqrstuvwxyz1234567890';
	var maxPos = $chars.length;
	var pwd = '';
	for (i = 0; i < len; i++) {
		pwd += $chars.charAt(Math.floor(Math.random() * maxPos));
	}
	return pwd;
};

function randomSecret() {
	var Token = document.getElementById("cbid.aria2.main.rpc_secret");
	Token.value = randomString(32);
};

function showRPCURL() {
	var websocket = document.getElementById("use_websocket");
	var protocol = (websocket.checked) ? "ws" : "http";
	var newTextNode = document.getElementById("aria2rpcpath");
	var auth_method = document.getElementById("cbid.aria2.main.rpc_auth_method");
	var auth_port = document.getElementById("cbid.aria2.main.rpc_listen_port");
	if (auth_port.value == "") {
		auth_port_value = "6800"
	} else {
		auth_port_value = auth_port.value
	};
	if (auth_method.value == "token") {
		var auth_token = document.getElementById("cbid.aria2.main.rpc_secret");
		newTextNode.value = protocol + "://token:" + auth_token.value + "@" + document.domain + ":" + auth_port_value + "/jsonrpc";
	} else if (auth_method.value == "user_pass") {
		var auth_user = document.getElementById("cbid.aria2.main.rpc_user");
		var auth_passwd = document.getElementById("cbid.aria2.main.rpc_passwd");
		newTextNode.value = protocol + "://" + auth_user.value + ":" + auth_passwd.value + "@" + document.domain + ":" + auth_port_value + "/jsonrpc";
	} else {
		newTextNode.value = protocol + "://" + document.domain + ":" + auth_port_value + "/jsonrpc";
	}
};

function openWebUI(path) {
	var curWwwPath = window.document.location.href;
	var pathName = window.document.location.pathname;
	var pos = curWwwPath.indexOf(pathName);
	var localhostPath = curWwwPath.substring(0, pos);
	var url = "http:" + localhostPath.substring(window.location.protocol.length) + "/" + path;
	window.open(url)
};
//]]>
</script>
<style>.mar-10 {margin-left: 10px; margin-right: 10px;}</style>
<fieldset class="cbi-section">
	<legend><%:Aria2 Status%></legend>
	<p id="aria2_status">
		<em><%:Collecting data...%></em>
	</p>
</fieldset>

```

hostkit, 2016-07-14:
this /usr/lib/lua/luci/view/aria2/web_script.htm and this is /usr/lib/lua/luci/controller/aria2.lua
[Documents.zip](https://github.com/openwrt/packages/files/363616/Documents.zip)

kuoruan, 2016-07-14:
You can try remove /usr/lib/lua/luci/view/aria2/web_script.htm, and see if it works well.

hostkit, 2016-07-14:
```
Thu Jul 14 12:10:20 2016 daemon.err uhttpd[1182]: /usr/lib/lua/luci/dispatcher.lua:433: Failed to execute cbi dispatcher target for entry '/admin/services/aria2'.
Thu Jul 14 12:10:20 2016 daemon.err uhttpd[1182]: The called action terminated with an exception:
Thu Jul 14 12:10:20 2016 daemon.err uhttpd[1182]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Thu Jul 14 12:10:20 2016 daemon.err uhttpd[1182]: A runtime error occured: /usr/lib/lua/luci/template.lua:74: Failed to load template 'aria2/web_script'.
Thu Jul 14 12:10:20 2016 daemon.err uhttpd[1182]: Error while parsing template '/usr/lib/lua/luci/view/aria2/web_script.htm':
Thu Jul 14 12:10:20 2016 daemon.err uhttpd[1182]: No such file or directory
Thu Jul 14 12:10:20 2016 daemon.err uhttpd[1182]: stack traceback:
Thu Jul 14 12:10:20 2016 daemon.err uhttpd[1182]: 	[C]: in function 'assert'
Thu Jul 14 12:10:20 2016 daemon.err uhttpd[1182]: 	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
Thu Jul 14 12:10:20 2016 daemon.err uhttpd[1182]: 	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>

```

[image]

kuoruan, 2016-07-14:
OK, could you please give me your /usr/lib/lua/luci/model/cbi/aria2.lua

kuoruan, 2016-07-14:
Try delete this line https://github.com/openwrt/luci/blob/master/applications/luci-app-aria2/luasrc/model/cbi/aria2.lua#L47 in /usr/lib/lua/luci/model/cbi/aria2.lua.

hostkit, 2016-07-14:
[aria2.zip](https://github.com/openwrt/packages/files/363671/aria2.zip)

hostkit, 2016-07-14:
what it's normal?
[image]

kuoruan, 2016-07-14:
It seems that there's something wrong with web_script.htm, but I replace all my file with yours, works well.

Did you try remove luci-app-aria2, aria2 delete /etc/config/aria2 and reinstall them?

hostkit, 2016-07-14:
ok. i done. not fix
```
Thu Jul 14 12:40:12 2016 daemon.err uhttpd[1178]: The called action terminated with an exception:
Thu Jul 14 12:40:12 2016 daemon.err uhttpd[1178]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Thu Jul 14 12:40:12 2016 daemon.err uhttpd[1178]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'aria2/web_script'.
Thu Jul 14 12:40:12 2016 daemon.err uhttpd[1178]: A runtime error occured: [string "/usr/lib/lua/luci/view/aria2/web_script.htm"]:1: attempt to call global 'url' (a nil value)
Thu Jul 14 12:40:12 2016 daemon.err uhttpd[1178]: stack traceback:
Thu Jul 14 12:40:12 2016 daemon.err uhttpd[1178]: 	[C]: in function 'assert'
Thu Jul 14 12:40:12 2016 daemon.err uhttpd[1178]: 	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
Thu Jul 14 12:40:12 2016 daemon.err uhttpd[1178]: 	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>
```

kuoruan, 2016-07-14:
eh, Please replase this line https://github.com/openwrt/luci/blob/master/applications/luci-app-aria2/luasrc/view/aria2/web_script.htm#L2 with
```XHR.poll(5, '<%=url('admin/services/aria2/status')%>', null,```
in /usr/lib/lua/luci/view/aria2/web_script.htm, thanks.

hostkit, 2016-07-14:
show same error, i think error because XHR.poll(5, '<%=url(...)%>', null 
system cant find url define

hostkit, 2016-07-14:
if this function remove service can open but some service disable, unable to get status,
```
XHR.poll(5, '<%=url([[admin]], [[services]], [[aria2]], [[status]])%>', null,
	function(x, data) {
		var tb = document.getElementById('aria2_status');
		if (data && tb) {
			if (data.running) {
				var links = '<em><%:The Aria2 service is running.%></em>';
				if (data.yaaw) {
					links += '<input class="cbi-button mar-10" type="button" value="<%:Open YAAW%>" onclick="openWebUI(\'yaaw\');" />';
				}
				if (data.webui) {
					links += '<input class="cbi-button mar-10" type="button" value="<%:Open WebUI-Aria2%>" onclick="openWebUI(\'webui-aria2\');" />';
				}
				tb.innerHTML = links;
			} else {
				tb.innerHTML = '<em><%:The Aria2 service is not running.%></em>';
			}
		}
	}
);

```

how to access webui? what your used?

kuoruan, 2016-07-14:
I am using webui-aria2, you can find it in packages.

kuoruan, 2016-07-14:
That's really strange. What's your Firmware Version?

hostkit, 2016-07-14:
webui-aria2 still same. mr3420 openwrt 15.05 exroot

kuoruan, 2016-07-14:
luci-app-aria2 have nothing to do with yaaw or webui-aria2. you can use either of them.

hostkit, 2016-07-14:
yeah i know, but what is happen? What's your Firmware Version?

kuoruan, 2016-07-14:
[web_script.zip](https://github.com/openwrt/packages/files/363808/web_script.zip)

try this file. just formatted the code...

hostkit, 2016-07-14:
not fix, same problem. in sub service, i dont have sub menu aria service, it is problem? only aria setting on sub menu service

kuoruan, 2016-07-14:
That's right.
[image]

kuoruan, 2016-07-14:
Replace 2L with
```XHR.poll(5, '/cgi-bin/luci/admin/services/aria2/status', null,```
use the real url. hope this can fix it.

hostkit, 2016-07-14:
```
/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute cbi dispatcher target for entry '/admin/services/aria2'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'aria2/web_script'.
A runtime error occured: [string "/usr/lib/lua/luci/view/aria2/web_script.htm"]:1: attempt to call global 'url' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function 
```

hostkit, 2016-07-14:
When access  /cgi-bin/luci/admin/services/aria2/status
[image]

kuoruan, 2016-07-15:
web_script.htm is like https://github.com/openwrt/luci/blob/master/applications/luci-app-minidlna/luasrc/view/minidlna_status.htm, can you install a minidlna to see if this works well?

hostkit, 2016-07-15:
fix with this:
Replace /usr/lib/lua/luci/view/aria2#L2
`XHR.poll(5, '<%=luci.dispatcher.build_url("admin", "services", "aria2", "status")%>', null,`

hostkit, 2016-07-15:
why service not running? 
config was enable, reboot not fix and startup aria2 enable and start

hostkit, 2016-07-15:
please test on your openwrt and update your script github.

hostkit, 2016-07-15:
please test on your openwrt and update your script github.
fix with this:
Replace /usr/lib/lua/luci/view/aria2/web_script.htm#L2
`XHR.poll(5, '<%=luci.dispatcher.build_url("admin", "services", "aria2", "status")%>', null,`

hostkit, 2016-07-15:
can  you update /etc/init.d/aria2 start stop restart like this, i think more awesome if you add logger in syslog
```
start() {
		logger -t ARIA2C 'Starting aria2c service'
		config_load 'aria2'
		config_foreach start_instance 'aria2'
	sleep 5
		return 0
}

stop() {
	if [ -n "`pidof aria2c`" ]; then
		logger -t ARIA2C 'Shutting down aria2c service'
		service_stop /usr/bin/aria2c
		sleep 5
	fi
	return 0
}

restart() {
	logger -t ARIA2C 'Restarting aria2s service'
	stop
	sleep 2
	start
}
```

closed 2016-07-15 by hostkit

kuoruan, 2016-07-15:
https://github.com/openwrt/luci/commit/281d2f617899a0c3a171a66c6a3a14e834aa000c
In this commit, replaced luci.dispatcher.build_url(...) with url(...), so I think luci.dispatcher.build_url is  deprecated? @jow-

kuoruan, 2016-07-16:
I think the ```sleep 5``` is unnessary, what' your opinion?

hostkit, 2016-07-16:
just give time break from next command :d

hostkit, 2016-07-16:
> In this commit, replaced luci.dispatcher.build_url(...) with url(...), so I think luci.dispatcher.build_url is deprecated? @jow- 
im use 15.05 but this command still here at some code,  What's your Firmware Version?

hostkit, 2016-07-16:
/usr/lib/lua/luci/view/admin_network/wifi_overview.htm#L170
[wifi_overview.zip](https://github.com/openwrt/packages/files/367509/wifi_overview.zip)
/usr/lib/lua/luci/view/admin_network/wifi_status.htm#L2
[wifi_status.zip](https://github.com/openwrt/packages/files/367510/wifi_status.zip)

hostkit, 2016-07-23:
@kuoruan can you fix this with
if url(...) == nil then
luci.dispatcher.build_url(...)

im newbie coding especially lua, so can you help me?
i want read lua variable /usr/lib/lua/luci/model/cbi/aria2.lua (result above function) in /usr/lib/lua/luci/view/aria2/web_script.htm

hostkit, 2016-07-24:
what this can fix nil url

> <%
> 	local aria2_path
> 	if url == nil then
> 		aria2_path = luci.dispatcher.build_url([[admin]], [[services]], [[aria2]], [[status]])
> 	else		
>   		aria2_path = url([[admin]], [[services]], [[aria2]], [[status]])
> 	end
> %>
> 

/usr/lib/lua/luci/view/aria2/web_script.htm add before frist line