openwrt/packages #2963: perl is missing arm64(aarch64) config

plntyk, 2016-07-13:
reported by buildbot: 
http://buildbot.openwrt.org:8010/broken_packages/arm64/perl/compile.txt

& replicated locally

```
Can't open files/aarch64.config: No such file or directory
```

codemarauder, 2018-01-25:
I am also facing the same problem while trying to build packages for Espressobin board.

pprindeville, 2018-01-25:
@plntyk Is this still reproducible?

@codemarauder Can you please give more detail, like actual cut&paste of build output?

pprindeville, 2018-01-25:
And I don't think that link is valid anymore.

tmn505, 2018-01-26:
@pprindeville The build fails because of: https://github.com/openwrt/packages/blob/74eecfebcc3acabd9f5a68b3d0291289ef28c3c8/lang/perl/Makefile#L107

@codemarauder Please test PR #5503.

codemarauder, 2018-01-26:
Hi,

Here are the logs:

make package/perl/compile V=s
make[1]: Entering directory '/data/openwrt/espressobin/tmn505/source'
make[2]: Entering directory '/data/openwrt/espressobin/tmn505/source/package/libs/toolchain'
echo "libc" >> /data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libstdcpp" >> /data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "librt" >> /data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/data/openwrt/espressobin/tmn505/source/package/libs/toolchain'
make[2]: Entering directory '/data/openwrt/espressobin/tmn505/source/package/libs/zlib'
echo "zlib" >> /data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/zlib.default.install
make[2]: Leaving directory '/data/openwrt/espressobin/tmn505/source/package/libs/zlib'
make[2]: Entering directory '/data/openwrt/espressobin/tmn505/source/feeds/packages/libs/libxml2'
make[2]: Leaving directory '/data/openwrt/espressobin/tmn505/source/feeds/packages/libs/libxml2'
make[2]: Entering directory '/data/openwrt/espressobin/tmn505/source/package/libs/uclibc++'
make[2]: Leaving directory '/data/openwrt/espressobin/tmn505/source/package/libs/uclibc++'
make[2]: Entering directory '/data/openwrt/espressobin/tmn505/source/feeds/packages/libs/db47'
make[2]: Leaving directory '/data/openwrt/espressobin/tmn505/source/feeds/packages/libs/db47'
make[2]: Entering directory '/data/openwrt/espressobin/tmn505/source/package/libs/gettext-full'
make[2]: Leaving directory '/data/openwrt/espressobin/tmn505/source/package/libs/gettext-full'
make[2]: Entering directory '/data/openwrt/espressobin/tmn505/source/feeds/packages/libs/gdbm'
make[2]: Leaving directory '/data/openwrt/espressobin/tmn505/source/feeds/packages/libs/gdbm'
make[2]: Entering directory '/data/openwrt/espressobin/tmn505/source/feeds/packages/lang/perl'
make[2]: Leaving directory '/data/openwrt/espressobin/tmn505/source/feeds/packages/lang/perl'
make[2]: Entering directory '/data/openwrt/espressobin/tmn505/source/feeds/packages/lang/perl'
rm -f /data/openwrt/espressobin/tmn505/source/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.26.1/.configured_*
rm -f /data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.perl_installed
/data/openwrt/espressobin/tmn505/source/staging_dir/hostpkg/usr/bin/perl5.26.0 files/perlconfig.pl -Dowrt:target_cc='aarch64-openwrt-linux-musl-gcc' -Dowrt:gccversion="5.5.0" -Dowrt:target_cross='aarch64-openwrt-linux-musl-' -Dowrt:cflags='-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -I/data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/include -I/data/openwrt/espressobin/tmn505/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-5.5.0_musl/usr/include -I/data/openwrt/espressobin/tmn505/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-5.5.0_musl/include/fortify -I/data/openwrt/espressobin/tmn505/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-5.5.0_musl/include' -Dowrt:ldflags='-L/data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -L/data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl/lib -L/data/openwrt/espressobin/tmn505/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-5.5.0_musl/usr/lib -L/data/openwrt/espressobin/tmn505/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-5.5.0_musl/lib -znow -zrelro' -Dowrt:libc="musl" -Dowrt:ipv6=undef -Dowrt:threads=no -Dowrt:staging_dir='/data/openwrt/espressobin/tmn505/source/staging_dir/target-aarch64_cortex-a53_musl' -Dowrt:host_perl_prefix='/data/openwrt/espressobin/tmn505/source/staging_dir/hostpkg/usr' files/version.config files/base.config files/aarch64.config files/architecture.config files/signal.config files/threads.config files/libc.config files/misc.config > /data/openwrt/espressobin/tmn505/source/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.26.1/config.sh
Can't open files/aarch64.config: No such file or directory
Makefile:141: recipe for target '/data/openwrt/espressobin/tmn505/source/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.26.1/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[2]: *** [/data/openwrt/espressobin/tmn505/source/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.26.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
make[2]: Leaving directory '/data/openwrt/espressobin/tmn505/source/feeds/packages/lang/perl'
package/Makefile:109: recipe for target 'package/feeds/packages/perl/compile' failed
make[1]: *** [package/feeds/packages/perl/compile] Error 2
make[1]: Leaving directory '/data/openwrt/espressobin/tmn505/source'
/data/openwrt/espressobin/tmn505/source/include/toplevel.mk:216: recipe for target 'package/perl/compile' failed
make: *** [package/perl/compile] Error 2

codemarauder, 2018-01-26:
I am going to start afresh as something seems to be not right. The Perl compiled fine after rebuilding musl but I see 2 types of arch in bin/packages now, namely:
* aarch64_cortex-a53  
* arm_cortex-a9_vfpv3

Perl ended up in arm_cortex-a9_vrpv3 with other packages in "base".

Will report in a few hours again.

codemarauder, 2018-01-26:
Perl is compiling successfully with following:
* LEDE Source from https://github.com/tmn505/source (branch espressobin)
* Packages from http://github.com/tmn505/packages.git^6535c763953416c5ff74469d8262c750413fa126

Thanks for the great help.

PR #5503 (2018-01-26): perl: add build config for aarch64
Because nonexistent aarch64.config the build of perl will fail for ARM
64-bit targets. Fix it by adding mentioned config. Fixes #2963.

The config.sh file was generated on ESPRESSObin with OpenWrt/master and perl 5.26.1 source.
Package tested on the same system.

Maintainer: @Naoir  @pprindeville
commit 6535c763 (2018-01-26): perl: add build config for aarch64

Because nonexistent aarch64.config the build of perl will fail for ARM
64-bit targets. Fix it by adding mentioned config. Fixes #2963.
closed 2018-01-26 by pprindeville