openwrt/packages #2978: Kismet - Error: redefinition of 'struct ethhdr'

kietnlt, 2016-07-21:
Dear everyone,
I compile kismet 2016-07-R1 on OpenWrt Trunk but I have some error:

In file included from /home/kiet/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.14/include/linux/if_tun.h:20:0,
                 from dumpfile_tuntap.h:68,
                 from dumpfile_tuntap.cc:24:
/home/kiet/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.14/include/linux/if_ether.h:140:8: error: redefinition of 'struct ethhdr'
 struct ethhdr {
        ^
In file included from ifcontrol.h:42:0,
                 from dumpfile_tuntap.cc:23:
/home/kiet/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.14/include/netinet/if_ether.h:96:8: error: previous definition of 'struct ethhdr'
 struct ethhdr {
        ^
Makefile:317: recipe for target 'dumpfile_tuntap.o' failed
make[4]: *** [dumpfile_tuntap.o] Error 1
make[4]: Leaving directory '/home/kiet/openwrt/build_dir/target-mips_34kc_musl-1.1.14/kismet-2016-07-R1'
Makefile:142: recipe for target '/home/kiet/openwrt/build_dir/target-mips_34kc_musl-1.1.14/kismet-2016-07-R1/.built' failed
make[3]: *** [/home/kiet/openwrt/build_dir/target-mips_34kc_musl-1.1.14/kismet-2016-07-R1/.built] Error 2
make[3]: Leaving directory '/home/kiet/openwrt/feeds/packages/net/kismet'
package/Makefile:196: recipe for target 'package/feeds/packages/kismet/compile' failed
make[2]: *** [package/feeds/packages/kismet/compile] Error 2
make[2]: Leaving directory '/home/kiet/openwrt'
package/Makefile:193: recipe for target '/home/kiet/openwrt/staging_dir/target-mips_34kc_musl-1.1.14/stamp/.package_compile' failed
make[1]: *** [/home/kiet/openwrt/staging_dir/target-mips_34kc_musl-1.1.14/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kiet/openwrt'
/home/kiet/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2

Any help would be much appreciated.

yousong, 2016-08-02:
ping kismet maintainer @sourceindex :)

thess, 2016-08-08:
Should be fixed in #3029

closed 2016-08-08 by thess