openwrt/packages #2981: Unable to compile uboot-envtools: unknown type name 'uint8_t'

michael-dev, 2016-07-21:
make[3]: Entering directory '/opt/openwrt/stage2/trunk-devel/build_dir/target-powerpc_8540_glibc-2.22/u-boot-2015.10'
  HOSTCC  tools/env/fw_env.o
tools/env/fw_env.c:57:2: error: unknown type name 'uint8_t'
  uint8_t mtd_type;  /* type of the MTD device */
  ^

michael-dev, 2016-07-21:
workaround:

cat ~/stage2/trunk-devel/package/boot/uboot-envtools/patches/400-fix-stdtype.patch
--- u-boot-2015.10/tools/env/fw_env.c.orig      2016-07-21 14:33:13.842378135 +0200
+++ u-boot-2015.10/tools/env/fw_env.c   2016-07-21 14:33:56.291010273 +0200
@@ -22,9 +22,9 @@
 #include <sys/ioctl.h>
 #include <sys/stat.h>
 #include <unistd.h>
+#include <stdint.h>
 
 #ifdef MTD_OLD
-# include <stdint.h>
 # include <linux/mtd/mtd.h>
 #else
 # define  __user       /* nothing */

yousong, 2016-08-02:
`uboot-envtools` is part of base packages, so please consider submitting the patch to the mailing list, either `openwrt-devel` or `lede-dev` and I think devs will try and pick it up.

closed 2016-08-02 by yousong

michael-dev, 2016-08-17:
Isn't this the repo for base packages?

yousong, 2016-08-17:
base packages are maintained along with build systems.  For LEDE, it's at https://git.lede-project.org/?p=source.git;a=tree;f=package/boot/uboot-envtools;h=6469bfc0141b22174a84313c12398d18d3533356;hb=HEAD