openwrt/packages #2985: aria2c 1.24 or 1.25 not running

hostkit, 2016-07-23:
Hello @kuoruan 
1. i have been installed aria2 1.25 or aria2 1.24 when i want to start aria2 /usr/bin/aria2c said not found but this file exist, if im replaced with older version work normally? can you help me?

2. after you modif /etc/init.d/aria2
```
+	option_disabled "$s" 'enable_log' || {
 +                touch "$log_file"
 +                echo -e "log=$log_file" >> $config_file
 +        }
```
/etc/init.d/aria2 not working, but if this line remove work normallly

kuoruan, 2016-07-24:
if you want to use aria2 log, you should change user to root.

kuoruan, 2016-07-24:
Ok, I well make a change, if chose root user, then can enable aria2 log.

hostkit, 2016-07-24:
for question number 1, why it's  happen?
working if aria2c repalce using aria2c from old version?

kuoruan, 2016-07-24:
Try chmod +x /usr/bin/aria2c

kuoruan, 2016-07-24:
By the way, do you have a QICQ or WeChat?

hostkit, 2016-07-24:
Only WhatsApp and bbm :)

hostkit, 2016-07-24:
not working chmod +x
[image]

kuoruan, 2016-07-24:
Where your package from?

hostkit, 2016-07-24:
download from downloads.openwrt.org snapshot, but my firmware 15.0.5, firmware 15.0.5.1 not working too. my connection very limited so unable build from source, i have vps but always build  fail. g++ kill selft :(

hostkit, 2016-07-24:
i always wait update from snapshot when you update it

kuoruan, 2016-07-24:
What's your device model, I can build one for you.

hostkit, 2016-07-24:
TLMR3420v1 standard

hostkit, 2016-07-24:
Its true step by step compile packages from source ?
> git clone https://git.openwrt.org/openwrt.git
> cd openwrt
> ./scripts/feeds update -a
> ./scripts/feeds install -a
> make menuconfig
> make tools/install
> make toolchain/install
> make package/aria2/{clean,compile,install} V=99 DEVELOPER=1

kuoruan, 2016-07-24:
You are using 15.05, you should get source use ```git clone git://git.openwrt.org/15.05/openwrt.git```

All version are here: https://dev.openwrt.org/wiki/GetSource

hostkit, 2016-07-24:
if i am use 15.0.5 can i compile your luci-aapp-aria2 and aria2 updated from github?
i think my vps very low :(

kuoruan, 2016-07-24:
I think they can use both in Chaos Calmer and Designated Driver.

kuoruan, 2016-07-25:
download this: https://soft.kuoruan.com/openwrt/ar71xx/packages/packages/aria2_1.25.0-2_ar71xx.ipk

hostkit, 2016-07-25:
ok aria2 work normally, why this happen?

kuoruan, 2016-07-26:
Maybe it's because OpenWrt switches to musl by default, replacing uClibc in Designated Driver.

hostkit, 2016-07-26:
how to change uclibc on image generator or must be use  source

kuoruan, 2016-07-26:
You mast use source to build it, and uClibc is removed in trunk.

closed 2016-07-30 by hostkit