openwrt/packages #2989: error to build someone - "missing dependencies libz.so.1"

build000, 2016-07-24:
https://github.com/lede-project/source/issues/213

nbd168, 2016-07-24:
Simply linking another ticket here and leaving it up to everybody else to figure out what's broken is just too lazy ;)

Please provide a proper description, as well as a list of packages that are still broken. By the way, I already pushed some fixes, so please make sure you update your tree and test again before providing further details.

build000, 2016-07-24:
At the moment, the updating of sources, all interesting to me to build packages:
```
sj@slawek:~/Pulpit/Projekty/LEDE-WNDR4700$ date
nie, 24 lip 2016, 15:13:49 CEST 
sj@slawek:~/Pulpit/Projekty/LEDE-WNDR4700$ make package/symlinks V=s
./scripts/feeds update -a
Updating feed 'packages' from 'https://github.com/openwrt/packages.git' ...
remote: Counting objects: 63, done.
remote: Compressing objects: 100% (33/33), done.
remote: Total 63 (delta 38), reused 52 (delta 27), pack-reused 0
Unpacking objects: 100% (63/63), done.
From https://github.com/openwrt/packages
   02ff460..fb37ae2  master     -> origin/master
Updating 02ff460..fb37ae2
Fast-forward
 libs/libtorrent/Makefile  |  2 +-
 libs/neon/Makefile        |  2 +-
 mail/mutt/Makefile        |  2 +-
 net/aircrack-ng/Makefile  |  2 +-
 net/chaosvpn/Makefile     |  2 +-
 net/freeradius2/Makefile  |  2 +-
 net/git/Makefile          |  2 +-
 net/stunnel/Makefile      |  2 +-
 net/tinc/Makefile         |  2 +-
 net/tor/Makefile          |  2 +-
 net/transmission/Makefile | 12 ++++++------
 net/wget/Makefile         |  3 +--
 12 files changed, 17 insertions(+), 18 deletions(-)
Create index file './feeds/packages.index' 
Collecting package info: done
Updating feed 'luci' from 'https://github.com/openwrt/luci.git' ...
Already up-to-date.
Create index file './feeds/luci.index' 
Updating feed 'routing' from 'https://github.com/openwrt-routing/packages.git' ...
Already up-to-date.
Create index file './feeds/routing.index' 
Updating feed 'telephony' from 'https://github.com/openwrt/telephony.git' ...
Already up-to-date.
Create index file './feeds/telephony.index' 
./scripts/feeds install -a
Collecting package info: done
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
```
How do I find any more to let you know for this issue.
Thx.
:dancer:

thess, 2016-08-05:
I see no errors here?

Original problem hould have been fixed by 3759a36711dd0327ffc1e7b79e0a44a9236c8bbb

closed 2016-08-05 by thess