openwrt/packages #2990: error build python-cryptography - probably last commits to openssl

build000, 2016-07-24:
```
(...)
make[3]: Entering directory '/home/sj/Pulpit/Projekty/LEDE-WNDR4300/feeds/packages/lang/python-cryptography'
install -d -m0755 /home/sj/Pulpit/Projekty/LEDE-WNDR4300/build_dir/target-mips_34kc_musl-1.1.15/cryptography-1.3.1/ipkg-install//usr/lib/python2.7/site-packages
if [ "" == "HOST" ]; then export PYTHONPATH="/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/host/lib/python2.7:/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/host//lib/python2.7/site-packages"; export PYTHONDONTWRITEBYTECODE=0; else export PYTHONPATH="/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/usr/lib/python2.7:/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15//usr/lib/python2.7/site-packages:/home/sj/Pulpit/Projekty/LEDE-WNDR4300/build_dir/target-mips_34kc_musl-1.1.15/cryptography-1.3.1/ipkg-install//usr/lib/python2.7/site-packages"; export PYTHONDONTWRITEBYTECODE=1; export _python_sysroot="/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15"; export _python_prefix="/usr"; export _python_exec_prefix="/usr"; fi; export PYTHONOPTIMIZE="";  cd /home/sj/Pulpit/Projekty/LEDE-WNDR4300/build_dir/target-mips_34kc_musl-1.1.15/cryptography-1.3.1/; CC="mips-openwrt-linux-musl-gcc" CCSHARED="mips-openwrt-linux-musl-gcc -fpic" CXX="mips-openwrt-linux-musl-g++" LD="mips-openwrt-linux-musl-gcc" LDSHARED="mips-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/sj/Pulpit/Projekty/LEDE-WNDR4300/build_dir/target-mips_34kc_musl-1.1.15/cryptography-1.3.1:cryptography-1.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -mno-mips16 -mno-interlink-mips16" CPPFLAGS="-I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/toolchain-mips_34kc_gcc-5.4.0_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/toolchain-mips_34kc_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/toolchain-mips_34kc_gcc-5.4.0_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/usr/include/python2.7" LDFLAGS="-L/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/usr/lib -L/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/lib -L/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/toolchain-mips_34kc_gcc-5.4.0_musl-1.1.15/usr/lib -L/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/toolchain-mips_34kc_gcc-5.4.0_musl-1.1.15/lib -znow -zrelro -lpython2.7" _PYTHON_HOST_PLATFORM=linux2 __PYVENV_LAUNCHER__="/usr/bin/python2.7"   /home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/host/bin/python2.7  ./setup.py install --prefix="/usr" --root="/home/sj/Pulpit/Projekty/LEDE-WNDR4300/build_dir/target-mips_34kc_musl-1.1.15/cryptography-1.3.1/ipkg-install" ;
running install
running build
running build_py
running egg_info
writing requirements to src/cryptography.egg-info/requires.txt
writing src/cryptography.egg-info/PKG-INFO
writing top-level names to src/cryptography.egg-info/top_level.txt
writing dependency_links to src/cryptography.egg-info/dependency_links.txt
writing entry points to src/cryptography.egg-info/entry_points.txt
reading manifest file 'src/cryptography.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
no previously-included directories found matching 'docs/_build'
warning: no previously-included files matching '*' found under directory 'vectors'
writing manifest file 'src/cryptography.egg-info/SOURCES.txt'
warning: build_py: byte-compiling is disabled, skipping.

running build_ext
generating cffi module 'build/temp.linux2-2.7/_padding.c'
already up-to-date
generating cffi module 'build/temp.linux2-2.7/_constant_time.c'
already up-to-date
generating cffi module 'build/temp.linux2-2.7/_openssl.c'
already up-to-date
building '_openssl' extension
mips-openwrt-linux-musl-gcc -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/sj/Pulpit/Projekty/LEDE-WNDR4300/build_dir/target-mips_34kc_musl-1.1.15/cryptography-1.3.1:cryptography-1.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -mno-mips16 -mno-interlink-mips16 -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/toolchain-mips_34kc_gcc-5.4.0_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/toolchain-mips_34kc_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/toolchain-mips_34kc_gcc-5.4.0_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/usr/include/python2.7 -fPIC -I/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/usr/include/python2.7 -c build/temp.linux2-2.7/_openssl.c -o build/temp.linux2-2.7/build/temp.linux2-2.7/_openssl.o
build/temp.linux2-2.7/_openssl.c:442:26: fatal error: openssl/cmac.h: No such file or directory
compilation terminated.
error: command 'mips-openwrt-linux-musl-gcc' failed with exit status 1
Makefile:53: recipe for target '/home/sj/Pulpit/Projekty/LEDE-WNDR4300/build_dir/target-mips_34kc_musl-1.1.15/cryptography-1.3.1/.built' failed
make[3]: *** [/home/sj/Pulpit/Projekty/LEDE-WNDR4300/build_dir/target-mips_34kc_musl-1.1.15/cryptography-1.3.1/.built] Error 1
make[3]: Leaving directory '/home/sj/Pulpit/Projekty/LEDE-WNDR4300/feeds/packages/lang/python-cryptography'
package/Makefile:187: recipe for target 'package/feeds/packages/python-cryptography/compile' failed
make[2]: *** [package/feeds/packages/python-cryptography/compile] Error 2
make[2]: Leaving directory '/home/sj/Pulpit/Projekty/LEDE-WNDR4300'
package/Makefile:184: recipe for target '/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/sj/Pulpit/Projekty/LEDE-WNDR4300/staging_dir/target-mips_34kc_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sj/Pulpit/Projekty/LEDE-WNDR4300'
/home/sj/Pulpit/Projekty/LEDE-WNDR4300/include/toplevel.mk:193: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
```

jefferyto, 2016-07-27:
This should be fixed by #2997.

build000, 2016-07-27:
```
(...)
make[3]: Entering directory '/home/sj/Pulpit/Projekty/LEDE-WNDR4700/feeds/packages/lang/python-cryptography'
install -d -m0755 /home/sj/Pulpit/Projekty/LEDE-WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/cryptography-1.4/ipkg-install//usr/lib/python2.7/site-packages
if [ "" == "HOST" ]; then export PYTHONPATH="/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7:/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/host//lib/python2.7/site-packages"; export PYTHONDONTWRITEBYTECODE=0; else export PYTHONPATH="/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/lib/python2.7:/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15//usr/lib/python2.7/site-packages:/home/sj/Pulpit/Projekty/LEDE-WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/cryptography-1.4/ipkg-install//usr/lib/python2.7/site-packages"; export PYTHONDONTWRITEBYTECODE=1; export _python_sysroot="/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15"; export _python_prefix="/usr"; export _python_exec_prefix="/usr"; fi; export PYTHONOPTIMIZE="";  cd /home/sj/Pulpit/Projekty/LEDE-WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/cryptography-1.4/; CC="powerpc-openwrt-linux-musl-gcc" CCSHARED="powerpc-openwrt-linux-musl-gcc -fPIC" CXX="powerpc-openwrt-linux-musl-g++" LD="powerpc-openwrt-linux-musl-gcc" LDSHARED="powerpc-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/sj/Pulpit/Projekty/LEDE-WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/cryptography-1.4:cryptography-1.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/python2.7" LDFLAGS="-L/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/lib -L/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/lib -L/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/lib -L/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/lib -znow -zrelro -lpython2.7" _PYTHON_HOST_PLATFORM=linux2 __PYVENV_LAUNCHER__="/usr/bin/python2.7"   /home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/host/bin/python2.7  ./setup.py install --prefix="/usr" --root="/home/sj/Pulpit/Projekty/LEDE-WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/cryptography-1.4/ipkg-install" ;
running install
running build
running build_py
running egg_info
writing requirements to src/cryptography.egg-info/requires.txt
writing src/cryptography.egg-info/PKG-INFO
writing top-level names to src/cryptography.egg-info/top_level.txt
writing dependency_links to src/cryptography.egg-info/dependency_links.txt
writing entry points to src/cryptography.egg-info/entry_points.txt
reading manifest file 'src/cryptography.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
no previously-included directories found matching 'docs/_build'
warning: no previously-included files matching '*' found under directory 'vectors'
writing manifest file 'src/cryptography.egg-info/SOURCES.txt'
warning: build_py: byte-compiling is disabled, skipping.

running build_ext
generating cffi module 'build/temp.linux2-2.7/_padding.c'
already up-to-date
generating cffi module 'build/temp.linux2-2.7/_constant_time.c'
already up-to-date
generating cffi module 'build/temp.linux2-2.7/_openssl.c'
already up-to-date
building '_openssl' extension
powerpc-openwrt-linux-musl-gcc -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/sj/Pulpit/Projekty/LEDE-WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/cryptography-1.4:cryptography-1.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/python2.7 -fPIC -I/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/python2.7 -c build/te [... cut ...]
package/Makefile:184: recipe for target '/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/sj/Pulpit/Projekty/LEDE-WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sj/Pulpit/Projekty/LEDE-WNDR4700'
/home/sj/Pulpit/Projekty/LEDE-WNDR4700/include/toplevel.mk:193: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
```
Unfortunately, there is no success - mentioned in the subject python module still does not compile.
Tested on LEDE-r1122.
:-1:

jefferyto, 2016-07-31:
The package compiles for me for your platform/target. Can you try compiling after a `make clean` or `make dirclean` (and updating your buildroot to trunk if necessary)?

build000, 2016-07-31:
of course it is - but the build error is still the same (Ubuntu 16.04.1 + current status of packages on the moment of writing this post).

jefferyto, 2016-08-01:
Can you post the content of these two files:
```
staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/openssl/opensslconf.h
staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/openssl/opensslv.h
```

Also, are r1122 and r1197 svn revisions? From where are you checking out the LEDE source code? AFAIK LEDE only has git repositories.

build000, 2016-08-01:
hmmm ... I do not understand - now compilation was successful without any problem - apparently undergone considerable changes since the sources Lede, which does not take into account subsequent updates sources of basic packages Lede and archive "packages" (apparently I had a mess in scripts sources).
Here are the compilation process from scratch, which is downloadable from new sources Lede (r1203), manual substitution "python-cryptography" to the version proposed by you (https://github.com/openwrt/packages/pull/2997), selecting only my architecture / router ( WNDR4700) without any additives except for python-cryptography and automatically selected to build relationships:
http://wklej.org/id/2776091/?hl=bash

Perhaps something causes problems with my additional packages from external sources and used the catalog "dl" has so posute packages that are built with an error - waiting for me, so a lot of work with the construction of the tree from scratch including my add-ons ...

yousong, 2016-08-02:
I am closing this one since the title says it's `SOLVED`.  Thank you for the time and efforts.

closed 2016-08-02 by yousong