openwrt/packages #2991: update URL for spdns is wrong, should be spdyn instead of spdns

pengel222, 2016-07-25:
Setting up ddns in OPenWRT for spdns I see warnings on spdns side:
Bitte verwenden Sie als Update-Server in Ihrem Client "update.spdyn.de" (Aktuelle Einstellung: update.spdns.de)
I believe the file /usr/lib/ddns/services needs to be changed.
Line
"spdns.de"	"http://[USERNAME]:[PASSWORD]@update.spdns.de/nic/update?hostname=[DOMAIN]&myip=[IP]"
to
"spdns.de"	"http://[USERNAME]:[PASSWORD]@update.**spdyn**.de/nic/update?hostname=[DOMAIN]&myip=[IP]"

chris5560, 2016-07-25:
Thanks for your advice.
Will be part of next ddns-script update.

diizzyy, 2017-06-09:
@chris5560 
Can we close this?

mhei, 2018-01-08:
I checked that this change was part of the update in #3134, so we can close this.

closed 2018-01-08 by mhei