openwrt/packages #3000: Cannot download libffi during compilation

DarthLabonne, 2016-07-28:
Hello everyone,

I've been trying to compile OpenWRT today by using the command `make -j1 V=s`

But somehow, every mirror for libffi-3.2.1 seem to be down (Connection timeout or 404) and compilation fails because of it.

I configured OpenWRT with "Advanced configuration options (for developers)", "Build the OpenWrt SDK" and Python 3 packages.

Can another mirror be set, do I just need to wait for mirrors to be back online again?

DarthLabonne, 2016-07-28:
Ok, so after searching for the problem, I found that the FTP links do work but my company's firewall was blocking the protocol.

As for the HTTP links, they ARE down, though.

yousong, 2016-08-02:
You can consider downloading the tarball from another place and do the verification of md5sum manually.  Place the tarball inside the download directory (usually `dl/`) and the build system should pick it up.

Feel free to reopen this issue if the above method does not work (it should...)

closed 2016-08-02 by yousong