openwrt/packages #3003: latest adblock (1.4.2) blocked ddns services - this is joke - right ?

build000, 2016-07-29:
https://github.com/openwrt/packages/commit/351babe8e38f392afd25b37b09114b9a60d9fc00 :

list "hphosts" - blocked freedns.afraid.org ddns services - very bad idea.
No effect its add services ddns (domain name/ip address) to /etc/adblock/adblock.whitelist.
Not possible to login this services and update address not properly (an address from space).

dibdot, 2016-07-29:
Put "afraid.org" in your whitelist and restart adblock (/etc/init.d/adblock restart) ... or don't use hphosts.

build000, 2016-07-29:
... Or let the author will fix the script - after I wrote that the addition of addresses / domain names with the service ddns afraid.org does not work - set for my domain (and others) shows the same address: 198.18.0.1, and the site simply you can not log in through the browser - the page is blocked.
:-1:

dibdot, 2016-07-29:
As I wrote before: no bug, nothing to fix! 
Simply repeat the steps described in my first answer and you can use afraid.org.

build000, 2016-07-29:
he ... how many times have you write that I added to the "whitelist" and it does not work ???
If does not work, it certainly is a script error, not a list that eg. In Windows, or different plug-ins for browsers that do not block freedns.afraid.org ...

dibdot, 2016-07-29:
Hugh? Whitelisting works as expected: 

**adblock with enabled hphosts and without "afraid.org" whitelist entry:**
<pre><code>
root@blackhole.nl:/tmp/dnsmasq.d$ grep "afraid.org" *
adb_list.hphosts:address=/afraid.org/198.18.0.1
adb_list.hphosts:address=/www.afraid.org/198.18.0.1
adb_list.shalla:address=/spookestreet.afraid.org/198.18.0.1
</pre></code>
<pre><code>
root@blackhole.nl:/tmp/dnsmasq.d$ nslookup afraid.org
Name:      afraid.org
Address 1: 198.18.0.1
</pre></code>

**adblock with enabled hphosts and with "afraid.org" whitelist entry and after adblock restart:**
<pre><code>
root@blackhole.nl:/tmp/dnsmasq.d$ grep "afraid.org" *
</pre></code>

<pre><code>
root@blackhole.nl:/tmp/dnsmasq.d$ nslookup afraid.org
Name:      afraid.org
Address 1: 204.140.20.21
Address 2: 2604:8b80:2:8::2
</pre></code>

No bug, nothing to fix!

build000, 2016-07-29:
For my:
**without "hphosts":**
```
root@WNDR4700:/tmp# cd dnsmasq.d
root@WNDR4700:/tmp/dnsmasq.d# grep "afraid.org" *
adb_list.shalla:address=/spookestreet.afraid.org/198.18.0.1
root@WNDR4700:/tmp/dnsmasq.d# nslookup afraid.org
nslookup: can't resolve '(null)': Name does not resolve

Name:      afraid.org
Address 1: 204.140.20.21
Address 2: 2604:8b80:2:8::2
root@WNDR4700:/tmp/dnsmasq.d#
```
**with "hphosts":**
```
root@WNDR4700:/tmp/dnsmasq.d# /etc/init.d/adblock restart
<13>Jul 29 16:01:50 adblock[19704] info : all adblock related services stopped
<13>Jul 29 16:01:50 adblock[19911] info : domain adblock processing started (1.4.2, r1158, 29.07.2016 16:01:50)
<13>Jul 29 16:01:50 adblock[19911] info : backup/restore will be disabled
<13>Jul 29 16:01:50 adblock[19911] info : created volatile firewall rulesets
<13>Jul 29 16:01:51 adblock[19911] info : created volatile uhttpd instances
<13>Jul 29 16:01:51 adblock[19911] info : => processing source 'adaway'
<13>Jul 29 16:01:52 adblock[19911] info :    source download finished (410 entries)
<13>Jul 29 16:01:52 adblock[19911] info :    domain merging finished
<13>Jul 29 16:01:52 adblock[19911] info : => processing source 'disconnect'
<13>Jul 29 16:01:54 adblock[19911] info :    source download finished (5709 entries)
<13>Jul 29 16:01:54 adblock[19911] info :    domain merging finished
<13>Jul 29 16:01:54 adblock[19911] info : => processing source 'dshield'
<13>Jul 29 16:01:56 adblock[19911] info :    source download finished (827 entries)
<13>Jul 29 16:01:56 adblock[19911] info :    domain merging finished
<13>Jul 29 16:01:56 adblock[19911] info : => processing source 'feodo'
<13>Jul 29 16:01:58 adblock[19911] info :    no online timestamp
<13>Jul 29 16:01:59 adblock[19911] info :    source download finished (0 entries)
<13>Jul 29 16:01:59 adblock[19911] info :    empty domain input, skipped
<13>Jul 29 16:01:59 adblock[19911] info : => processing source 'hphosts'
<13>Jul 29 16:02:40 adblock[19911] info :    source download finished (390478 entries)
<13>Jul 29 16:03:14 adblock[19911] info :    domain merging finished
<13>Jul 29 16:03:14 adblock[19911] info : => processing source 'malware'
<13>Jul 29 16:03:19 adblock[19911] info :    source download finished (24164 entries)
<13>Jul 29 16:03:22 adblock[19911] info :    domain merging finished
<13>Jul 29 16:03:22 adblock[19911] info : => processing source 'malwarelist'
<13>Jul 29 16:03:23 adblock[19911] info :    source download finished (1212 entries)
<13>Jul 29 16:03:23 adblock[19911] info :    domain merging finished
<13>Jul 29 16:03:23 adblock[19911] info : => processing source 'openphish'
<13>Jul 29 16:03:24 adblock[19911] info :    source download finished (3693 entries)
<13>Jul 29 16:03:24 adblock[19911] info :    domain merging finished
<13>Jul 29 16:03:24 adblock[19911] info : => processing source 'palevo'
<13>Jul 29 16:03:25 adblock[19911] info :    no online timestamp
<13>Jul 29 16:03:26 adblock[19911] info :    source download finished (14 entries)
<13>Jul 29 16:03:26 adblock[19911] info :    domain merging finished
<13>Jul 29 16:03:26 adblock[19911] info : => processing source 'ransomware'
<13>Jul 29 16:03:28 adblock[19911] info :    source download finished (450 entries)
<13>Jul 29 16:03:28 adblock[19911] info :    domain merging finished
<13>Jul 29 16:03:28 adblock[19911] info : => processing source 'rolist'
<13>Jul 29 16:03:40 adblock[19911] info :    source download finished (612 entries)
<13>Jul 29 16:03:40 adblock[19911] info :    domain merging finished
<13>Jul 29 16:03:40 adblock[19911] info : => processing source 'ruadlist'
<13>Jul 29 16:03:51 adblock[19911] info :    source download finished (2195 entries)
<13>Jul 29 16:03:52 adblock[19911] info :    domain merging finished
<13>Jul 29 16:03:52 adblock[19911] info : => processing source 'shalla'
<13>Jul 29 16:04:03 adblock[19911] info :    source download finished (32867 entries)
<13>Jul 29 16:04:06 adblock[19911] info :    domain merging finished
<13>Jul 29 16:04:06 adblock[19911] info : => processing source 'spam404'
<13>Jul 29 16:04:07 adblock[19911] info :    no online timestamp
<13>Jul 29 16:04:07 adblock[19911] info :    source download finished (5988 entries)
<13>Jul 29 16:04:08 adblock[19911] info :    domain merging finished
<13>Jul 29 16:04:08 adblock[19911] info : => processing source 'sysctl'
<13>Jul 29 16:04:10 adblock[19911] info :    source download finished (21195 entries)
<13>Jul 29 16:04:12 adblock[19911] info :    domain merging finished
[... 9 lines cut ...]
<13>Jul 29 16:04:22 adblock[19911] info :    domain merging finished
<13>Jul 29 16:04:22 adblock[19911] info : => processing source 'yoyo'
<13>Jul 29 16:04:23 adblock[19911] info :    source download finished (2381 entries)
<13>Jul 29 16:04:24 adblock[19911] info :    domain merging finished
<13>Jul 29 16:04:24 adblock[19911] info : => processing source 'zeus'
<13>Jul 29 16:04:25 adblock[19911] info :    source download finished (432 entries)
<13>Jul 29 16:04:25 adblock[19911] info :    domain merging finished
<13>Jul 29 16:04:25 adblock[19911] info : remove duplicates in separate block lists
<13>Jul 29 16:05:02 adblock[19911] info : block lists with overall 467740 domains loaded
<13>Jul 29 16:05:02 adblock[19911] info : firewall statistics (IPv4/IPv6): 0%/0% of all packets in prerouting chain are ad related & blocked
<13>Jul 29 16:05:02 adblock[19911] info : domain adblock processing finished successfully (1.4.2, r1158, 29.07.2016 16:05:02)
[... cut ...]  "afraid.org" *
adb_list.hphosts:address=/afraid.org/198.18.0.1
adb_list.hphosts:address=/www.afraid.org/198.18.0.1
adb_list.shalla:address=/spookestreet.afraid.org/198.18.0.1
root@WNDR4700:/tmp/dnsmasq.d# nslookup afraid.org
nslookup: can't resolve '(null)': Name does not resolve

Name:      afraid.org
Address 1: 198.18.0.1
root@WNDR4700:/tmp/dnsmasq.d# 
```
In both cases afraid.org added to the "whitelist"
So certainly it does not work the same as in you ...
In short - "afraid.org" is blocked.

dibdot, 2016-07-29:
With whitelisting you've got the same results than me, no "afraid.org" in the block lists and nslookup receives the answers from the orginal servers  ... I think you have restart/re-register the ddns services afterwards, but I don't use this package.

Still, nothing todo on adblock side.

build000, 2016-07-31:
1.4.3  - minimal positive adjustment:

This time does not block http (s): //freedns.afraid.org (I login to the site), and taking into account logs ddns and that the IP address of the "orange" changes every restart connection - you can see that it is correctly connection with the service and update addresses.
However, it is still blocked at the router login page luci from the outside, which is generally the address of the router / ISP address is blocked and luci displays the same as before with every domain name: 198.18.0.1.
I have the boil, that is progress but still connect to the domain, you can not.

dibdot, 2016-07-31:
As I wrote before: there is nothing to fix. 
1.4.3 addresses only some issues in restricted mode - nothing more. 
If you still have problems with the hphosts source, find out the relevant domains for whitelisting or simply don't use this source (it's disabled by default).

build000, 2016-07-31:
But something changed for the better - because now I can even log on to the site, so, however, something's up - a shame that no one has not taken another, which also benefits from this service, because I feel that you need more evidence that Your adblok the recent changes is simply "spierdolony" ... that's what.

yousong, 2016-08-02:
I think the issue of site being incorrectly blocked is at the moment solved so I am going to close this one.  Thanks for the feedback and efforts.

closed 2016-08-02 by yousong