openwrt/packages #3014: python-mini package should include the termios module

InsaneJ, 2016-08-02:
This issue was raised several years ago but still seems relevant:
https://dev.openwrt.org/ticket/6897
The package pyserial only requires the python-mini package rather than the full python package. Pyserial requires the python termios module, which is not included in the python-mini package. This prevents pyserial from being usable without the full python package. The termios module should be included in python-mini so that pyserial will be able to work without requiring the full python package.

Steps to reproduce:
Install python-mini and pyserial
Start the python interpreter and type "import serial"

> Python 2.6.4 (r264:75706, Mar 23 2010, 22:31:07)
> [GCC 4.3.3] on linux2
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import serial
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/usr/lib/python2.6/site-packages/serial/__init__.py", line 20, in <module>
>     from serialposix import *
>   File "/usr/lib/python2.6/site-packages/serial/serialposix.py", line 13, in <module>
>     import sys, os, fcntl, termios, struct, select, errno
> ImportError: No module named termios

Tested using:
python-mini_2.6.4-3_ar71xx
pyserial_2.4-1_ar71xx

commodo, 2016-08-02:
However now, the official Python package is at version 2.7.11
That version is unsupported.

commodo, 2017-05-26:
@InsaneJ 
can we close this issue ?
or is there anything else ?

the python version you mentioned is not supported by this repo [ and it's pretty old ]

InsaneJ, 2017-05-27:
You can close this issue if you want. After 10 months I wasn't expecting anything form this issue anymore.

closed 2017-05-27 by hnyman