openwrt/packages #3016: Can't connect to crelay built in http interface

szygmunt, 2016-08-02:
I can't connect to crelay http interface. I configured IP and port in /etc/crelay.conf. Start crelay by init script and scan IP with nmap - port defined in configuration file is opened but i can't connect.
I'm running CC 15.01 on x86 fujitsu thinclient.

thess, 2016-08-02:
What error do you get from your browser?

Could there be a firewall issue on the port and interface you attempting to connect to?

szygmunt, 2016-08-02:
I get "connection refused" in elinks and "unable to connect" in firefox. I don't think so that is firewall issue because i have http interface for transmission on this machine and it work good - i can connect to it from computers in lan. For a test i ran crelay on my PC and have no problem to connect to web interface. 
Thank you for interest.

thess, 2016-08-02:
Could you post the output of (on your device):

$ ifconfig

$ netstat -ltnp

$ cat /etc/crelay.conf

TIA

szygmunt, 2016-08-03:
ifconfig:
`eth0  Link encap:Ethernet  HWaddr 00:xx:xx:xx:xx:xx  
          inet addr:192.168.1.210  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:48629430 errors:0 dropped:0 overruns:0 frame:0
          TX packets:49201682 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:32263603367 (30.0 GiB)  TX bytes:49083877434 (45.7 GiB)
`
netstat -lntp

`Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:8000            0.0.0.0:*               LISTEN      24584/crelay
tcp        0      0 0.0.0.0:9091            0.0.0.0:*               LISTEN      1821/transmission-d
tcp        0      0 0.0.0.0:6566            0.0.0.0:*               LISTEN      1532/xinetd
tcp        0      0 0.0.0.0:8200            0.0.0.0:*               LISTEN      1505/minidlna
tcp        0      0 0.0.0.0:139             0.0.0.0:*               LISTEN      1568/smbd
tcp        4      0 0.0.0.0:9100            0.0.0.0:*               LISTEN      1509/p9100d
tcp        0      0 127.0.0.1:1234          0.0.0.0:*               LISTEN      9909/dropbear
tcp        0      0 0.0.0.0:51413           0.0.0.0:*               LISTEN      1821/transmission-d
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      2753/dropbear
tcp        0      0 0.0.0.0:9981            0.0.0.0:*               LISTEN      1851/tvheadend
tcp        0      0 0.0.0.0:445             0.0.0.0:*               LISTEN      1568/smbd
tcp        0      0 0.0.0.0:9982            0.0.0.0:*               LISTEN      1851/tvheadend
`
cat /etc/crelay.conf
`################################################
#
# crelay config file
#
# This file is read by crelay in daemon mode
# from /etc/crelay.conf
#
################################################

# HTTP server parameters
################################################
[HTTP server]
server_iface = 0.0.0.0    # listen interface IP address
#server_iface = 127.0.0.1 # to listen on localhost only
server_port  = 8000       # listen port
relay1_label = Device 1   # label for relay 1
relay2_label = Device 2   # label for relay 2
#relay3_label = Device 3   # label for relay 3
#relay4_label = Device 4   # label for relay 4
#relay5_label = Device 5   # label for relay 5
#relay6_label = Device 6   # label for relay 6
#relay7_label = Device 7   # label for relay 7
#relay8_label = Device 8   # label for relay 8
    
# GPIO driver parameters
################################################
[GPIO drv]
#num_relays = 8    # Number of GPIOs connected to relays (1 to 8)
#relay1_gpio_pin = 17   # GPIO pin for relay 1 (17 for RPi GPIO0)
#relay2_gpio_pin = 18   # GPIO pin for relay 2 (18 for RPi GPIO1)
#relay3_gpio_pin = 27   # GPIO pin for relay 3 (27 for RPi GPIO2)
#relay4_gpio_pin = 22   # GPIO pin for relay 4 (22 for RPi GPIO3)
#relay5_gpio_pin = 23   # GPIO pin for relay 5 (23 for RPi GPIO4)
#relay6_gpio_pin = 24   # GPIO pin for relay 6 (24 for RPi GPIO5)
#relay7_gpio_pin = 25   # GPIO pin for relay 7 (25 for RPi GPIO6)
#relay8_gpio_pin = 4    # GPIO pin for relay 8 ( 4 for RPi GPIO7)
    
# Sainsmart driver parameters
################################################
[Sainsmart drv]
num_relays = 4   # Number of relays on the Sainsmart card (4 or 8)
`

thess, 2016-08-03:
One last thing to check out... Try running crelay on the console so we can see if it is complaining about anything

$ /etc/init.d/crelay stop

$ crelay -d

You can also try interactive mode (cmd-line control) using
$ crelay -i

AFAIK - crelay has not been tested on 15.05. It is build on the current LEDE trunk and OpenWrt DD only. Are you building your own version?

szygmunt, 2016-08-03:
Hi,
running "crelay -d" and "telnet localhost 8000" in another terminal give me nothing. It means crelay doesn't deamonize as expect and telnet looks like hangs, send "Connection closed by foreign host" on Enter hit and quit.
crelay -i:
Detected relay card type is HID API compatible relay card (on 0002:0004:00, 2 channels)

I get crelay from trunk packages repository. Probably it is a problem. I can't compile my own version for 15.05 with errors like this:
make[3]: Entering directory `/home/szym/openwrt/build_dir/target-i386_geode_uClibc-0.9.33.2/hidapi-0.8.0-rc1/libusb'
make[3]: `libhidapi-libusb.so' is up to date.
make[3]: Leaving directory `/home/szym/openwrt/build_dir/target-i386_geode_uClibc-0.9.33.2/hidapi-0.8.0-rc1/libusb'
mv /home/szym/openwrt/build_dir/target-i386_geode_uClibc-0.9.33.2/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so /home/szym/openwrt/build_dir/target-i386_geode_uClibc-0.9.33.2/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so.0
mv: '/home/szym/openwrt/build_dir/target-i386_geode_uClibc-0.9.33.2/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so' and '/home/szym/openwrt/build_dir/target-i386_geode_uClibc-0.9.33.2/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so.0' are the same file
make[2]: *** [/home/szym/openwrt/build_dir/target-i386_geode_uClibc-0.9.33.2/hidapi-0.8.0-rc1/.built] Error 1
make[2]: Leaving directory `/home/szym/openwrt/feeds/packages/libs/hidapi'
make[1]: *** [package/feeds/packages/hidapi/compile] Error 2
make[1]: Leaving directory `/home/szym/openwrt'

szygmunt, 2016-08-04:
Managed to compile crelay on my own under 15.05 compile env. (did package/hidapi/clean before compiling crelay) but still have a problem to connect to http interface. Is this problem under trunk as well?

yousong, 2016-08-05:
Hi, @szygmunt 

From the output `netstat -lntp`, it looks like crelay listens on port 8000 just fine.  Were you tring to connect from lan or wan to port 8000?  Default firewall rules of OpenWrt will reject TCP INPUT connection from wan network with tcp-rst resulting in "connection refused" errmsg.

szygmunt, 2016-08-05:
I were trying 2 scenario:
1. connect from another computer in lan to daemon running on openwrt (no luck)
2. connect to http://localhost:8000 using elinks from openwrt machine running daemon itself (no luck as well)
Can someone confirm that crelay running under trunk has same problem with http interface?

thess, 2016-08-05:
I have tested crelay on a WNDR4300 using the LEDE trunk when you first reported this problem. Connecting to the HTTP interface on port 8000 displayed the UI with no problems. I will try crelay on 15.05 this weekend and report back what I find. I don't have an x86 platform available for testing.

Other possibilities are to move crelay to a different port in the unlikely case there is some issue with the other daemons you are running. Binding crelay to an explict IP/interface may also help. Also try running crelay alone without smbd, transmission, tvheadend, xinetd and minidlna.

thess, 2016-08-05:
OK - It fails on 15.05 but not on the latest trunk. I should have a fix for this by tomorrow. I'll supply more details after fix tested.

szygmunt, 2016-08-06:
Thank you. Btw. I experience the same behaviour on 15.05.1 on x86.

thess, 2016-08-06:
Attached is a patch to crelay for 15.05. I think there is something funky, in general, using file streams on a socket and that seems to be the problem. My guess is the problem you are experiencing is really a glitch in uClibc and it appears that musl does not display that symptom. I would be happier if crelay did not use file streams for network socket I/O (@ondrej1024 - please note).

This patch separates the file streams for input and output which seems to fix the connect/data truncation problem. I am not going to include this patch in the current trunk package until ondrej has a chance to consider a proper fix as I see this as a temporary hack.

[040-split_inout_streams.patch.txt](https://github.com/openwrt/packages/files/405194/040-split_inout_streams.patch.txt)

ondrej1024, 2016-08-07:
@thess, I'm just back from vacation and I will have a look at this asap (but can't promise that it will be quick). 
The web server code is mostly _borrowed_ from a simple web server implementation I found on the Internet. I didn't think about it too much as it seemed to work fine.

szygmunt, 2016-08-08:
@thess thank you for a patch - now crelay web interface is working. 
Ps. how to make a request for new packages? Actually i found Makefile for openwrt but it require patch I think.

thess, 2016-08-08:
If you are going to request a package to be created you should post the request under **Issues** in this repo (or the OpenWrt forums) and see if someone wants to pick it up and do it. If you have a package to submit and are willing to be the package maintainer, then read and follow our Guidelines here: https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md

If you are looking for help with the format and construction of the package, I suggest you post your work in the OpenWrt forum - You will get more help there.

closed 2016-08-08 by thess