openwrt/packages #3023: ffmpeg failed to compile

gendouhydeist, 2016-08-05:
@thess failed to compile on latest commit #3006 

> rm -f /home/Username/lede/source/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.ffmpeg.full_installed
( cd /home/Username/lede/source/build_dir/target-mips_24kc_musl-1.1.15/ffmpeg-full/ffmpeg-2.8.7; CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/Username/lede/source/build_dir/target-mips_24kc_musl-1.1.15/ffmpeg-full/ffmpeg-2.8.7:ffmpeg-2.8.7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/Username/lede/source/staging_dir/target-mips_24kc_musl-1.1.15/usr/include -I/home/Username/lede/source/staging_dir/target-mips_24kc_musl-1.1.15/include -I/home/Username/lede/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/usr/include -I/home/Username/lede/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/Username/lede/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include -fpic" LDFLAGS="-L/home/Username/lede/source/staging_dir/target-mips_24kc_musl-1.1.15/usr/lib -L/home/Username/lede/source/staging_dir/target-mips_24kc_musl-1.1.15/lib -L/home/Username/lede/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/usr/lib -L/home/Username/lede/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/lib -znow -zrelro" ./configure --enable-cross-compile --cross-prefix="mips-openwrt-linux-musl-" --arch="mips" --target-os=linux --prefix="/usr" --pkg-config="pkg-config" --enable-shared --enable-static --enable-small --enable-pthreads --enable-zlib --disable-runtime-cpudetect --disable-doc --disable-debug --enable-gpl --enable-version3 --disable-altivec --disable-amd3dnow --disable-amd3dnowext --disable-mmx --disable-mmxext --disable-sse --disable-sse2 --disable-sse3 --disable-ssse3 --disable-sse4 --disable-sse42 --disable-avx --disable-xop --disable-fma3 --disable-fma4 --disable-avx2 --disable-yasm --disable-inline-asm --disable-mips32r2 --disable-mipsdspr1 --disable-mipsdspr2 --disable-mipsfpu --disable-dxva2 --disable-lzma --disable-vaapi --disable-vda --disable-vdpau --disable-outdevs --disable-vfp --disable-neon --enable-libopus --enable-decoder=libopus --enable-libspeex --enable-decoder=libspeex --enable-libx264 --enable-libmp3lame )
ERROR: libmp3lame >= 3.98.3 not found

> If you think configure made a mistake, make sure you are using the latest
version from Git.  If the latest version fails, report the problem to the
ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.freenode.net.
Include the log file "config.log" produced by configure as this will help
solve the problem.
Makefile:625: recipe for target '/home/Username/lede/source/build_dir/target-mips_24kc_musl-1.1.15/ffmpeg-full/ffmpeg-2.8.7/.configured_nnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnyyyynyy' failed
make[3]: *** [/home/Username/lede/source/build_dir/target-mips_24kc_musl-1.1.15/ffmpeg-full/ffmpeg-2.8.7/.configured_nnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnyyyynyy] Error 1
make[3]: Leaving directory '/home/Username/lede/source/feeds/packages/multimedia/ffmpeg'
package/Makefile:124: recipe for target 'package/feeds/packages/ffmpeg/compile' failed
make[2]: *** [package/feeds/packages/ffmpeg/compile] Error 2
make[2]: Leaving directory '/home/Username/lede/source'
package/Makefile:121: recipe for target '/home/Username/lede/source/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/Username/lede/source/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/Username/lede/source'
/home/Username/lede/source/include/toplevel.mk:193: recipe for target 'world' failed
make: *** [world] Error 2

thess, 2016-08-05:
As the message says... Methinks you do not have the **lame** package installed in your buildroot.

`$ ./scripts/feeds install lame`

should fix it.

gendouhydeist, 2016-08-07:
confirmed and close

closed 2016-08-07 by gendouhydeist

artynet, 2017-02-10:
@gendouhydeist @thess 

the problem still persists for me ! Experienced on the latest trunks of OpenWRT and LEDE as well (libffmpeg-full is selected)...

thess, 2017-02-10:
@artynet - this issue is closed. Do not try to continue it with another problem. Please open a new issue and please, please supply some useful information about your environment and configuration, The output showing the failure excerpted from:
> make package/ffmpeg/{clean,compile} V=s

... would help as well.

Since we do not see any obvious failures in the buildbots (I haven't checked ALL targets) or in my test environment, your problem is more than likely in your environment and/or config options.