openwrt/packages #3040: lua-i2c: Upstream project no longer exist

Diftraku, 2016-08-09:
The upstream project for lua-i2c no longer exists on Scourge Forge: https://sourceforge.net/projects/i2c-lua/ returns a 404.

The project has been migrated to Github and lives at: https://github.com/mrpace2/i2c-lua

Diff between the original i2c-lua.c and the one from the repo are identical, as are the md5 sums.
```
$ diff i2c-lua.c.orig i2c-lua.c
$ md5sum i2c-lua.c*
bee4462f6d55dc47735495f65076d014  i2c-lua.c
bee4462f6d55dc47735495f65076d014  i2c-lua.c.orig
```
Up to date [i2c-lua-1.0.0.tar.gz](https://github.com/openwrt/packages/files/409035/i2c-lua-1.0.0.tar.gz) which has an md5 sum of `7098577f101410b201cd43a7467e408d` that matches the one listed in the Makefile.

hnyman, 2016-08-23:
Feel free to submit a pull request for updating the i2c-lua source download address from SF to Github.

Or maybe the package maintainer @mrpace2 will do that?

mrpace2, 2016-08-23:
Pull request #3087 should solve this issue.

mrpace2, 2016-08-23:
See #3089

hnyman, 2016-08-23:
upstream location change committed to master and for-15.05 branches.

closed 2016-08-23 by hnyman

ec1oud, 2018-11-30:
Now https://github.com/mrpace2/i2c-lua is also gone, but I found it here https://github.com/LuaDist2/i2c