openwrt/packages #3041: libx264: does no longer build, at least on x86_64 target

dibdot, 2016-08-09:
@ianchi 

libx264 no longer build on x86_64 target, see latest lede buildlogs [here](https://downloads.lede-project.org/snapshots/faillogs/x86_64/packages/libx264/compile.txt)

depend packages like ffmpeg & minidlna are broken, too.

thanks for looking into this.

ianchi, 2016-08-17:
The problem is that x86 targets need yasm to compile.
It seems that it is so even if you set "--disable-asm".

@thess perhaps while we try to solve this we should leave x264 out of ffmpeg-full for x86, so those packages don't break due to the dependency on x264

dangowrt, 2016-08-19:
I added yasm to the toolchain for x86 targets in https://github.com/lede-project/source/commit/c08651226f5645204f91f79d247801033f6015d1

thess, 2016-08-19:
Folks - I pushed this fix along with a minor CFLAGS cleanup. #3082.
ffmpeg cleanup follows

closed 2016-08-19 by thess

dibdot, 2016-08-20:
I've build & run tested the fixes above successfully on x86_64 - thanks!