openwrt/packages #3065: Start without update database and schedule update

hostkit, 2016-08-15:
@dibdot  how to start adblock without update source block ads?
Can you add this menu and schedule update every xxx hour xxx minutes

hostkit, 2016-08-15:
Sometimes my connection lost and reconnect again so my adblock restart again+update database, my quota very limited so can you helpme. Create menu start without update and diferent menu for update

dibdot, 2016-08-15:
@hostkit 
simply remove /etc/hotplug.d/iface/99-adblock and add an appropriate cron entry for adblock (see adblock readme.md for an example)

hostkit, 2016-08-15:
Ok thanks

closed 2016-08-15 by hostkit

dibdot, 2016-08-16:
#3071 
@hostkit in adblock version 1.4.5 I've added an option to fine tune the hotplug support, see new parm 'adb_hotplugif' in adblock readme.md)