openwrt/packages #3110: mwan3 - handling of ncm proto

lollul, 2016-08-30:
Maintainer: Jeroen Louwes <jeroen.louwes@gmail.com>
Environment: LEDE r1444, Linksys EA4500

Description: I'm using mwan3 with one plain ethernet/dhcp interface and second Huawei E3372 with ncm proto. ncm proto is using $INTERFACE_4 / $INTERFACE_6 as dynamic interfaces where ip and gateway are assigned. This is a problem for mwan3 as it is then not able to find the ip and gateway when it uses network_get_ipaddr and network_get_gateway functions.

I've applied the following patches to make it work. They are only for ipv4. I guess something similar should be done for ipv6 but at the same time I'm sure this problem can be solved much better... or maybe it is LEDE issue.

```
--- mwan3.sh_org        2016-08-28 22:35:11.187322358 +0200
+++ mwan3.sh    2016-08-28 22:35:11.180655688 +0200
@@ -122,16 +122,22 @@
 
 mwan3_create_iface_iptables()
 {
-       local id family src_ip src_ipv6
+       local id family src_ip src_ipv6 if_main if_proto
 
        config_get family $1 family ipv4
        mwan3_get_iface_id id $1
 
        [ -n "$id" ] || return 0
+       network_get_protocol if_proto $1
 
        if [ "$family" == "ipv4" ]; then
+               if [ "$if_proto" == "ncm" ]; then
+                       if_main=$1"_4"
+               else
+                       if_main=$1
+               fi
 
-               network_get_ipaddr src_ip $1
+               network_get_ipaddr src_ip $if_main
 
                $IPS -! create mwan3_connected list:set
 
@@ -231,16 +237,22 @@
 
 mwan3_create_iface_route()
 {
-       local id route_args
+       local id route_args if_proto if_main
 
        config_get family $1 family ipv4
        mwan3_get_iface_id id $1
 
        [ -n "$id" ] || return 0
 
+       network_get_protocol if_proto $1
        if [ "$family" == "ipv4" ]; then
-
-               network_get_gateway route_args $1
+                if [ "$if_proto" == "ncm" ]; then
+                        if_main=$1"_4"
+                else
+                        if_main=$1
+                fi
+       
+               network_get_gateway route_args $if_main
                route_args="via $route_args dev $2"
 
                $IP4 route flush table $id
```
```
--- 15-mwan3_org        2016-08-29 15:42:45.727976375 +0200
+++ 15-mwan3    2016-08-29 15:42:02.888413194 +0200
@@ -22,14 +22,21 @@
 [ -x /usr/sbin/ip6tables ] || exit 7
 [ -x /usr/bin/logger ] || exit 8
 
-local family gateway
+local family gateway if_proto if_main
 
 config_get family $INTERFACE family ipv4
+network_get_protocol if_proto $INTERFACE
+
+if [ "$if_proto" == "ncm" ]; then
+       if_main=$INTERFACE"_4"
+else
+       if_main=$INTERFACE
+fi
 
 if [ "$family" == "ipv4" ]; then
-       network_get_gateway gateway $INTERFACE
+       network_get_gateway gateway $if_main
 elif [ "$family" == "ipv6" ]; then
-       network_get_gateway6 gateway $INTERFACE
+       network_get_gateway6 gateway $if_main
 fi
 
 [ -n "$gateway" ] || exit 9
```

hnyman, 2016-08-30:
cc to maintainer: @Adze1502

ThomasCr, 2017-03-13:
thanks for your work.. ipv4 mwan3 works now for me too.
I made a updated diff, which applies cleanly on lede reboot.
But ipv6 is still not working. maybe not related to ncm proto. But if I stop mwan3 ipv6 works again.

[mwan3-ncm-proto-support.diff.txt](https://github.com/openwrt/packages/files/838547/mwan3-ncm-proto-support.diff.txt)

ThomasCr, 2017-03-13:
ok, I found the problem and updated the patch for ipv6.

[mwan3-ncm-proto-support.diff.txt](https://github.com/openwrt/packages/files/838577/mwan3-ncm-proto-support.diff.txt)

ThomasCr, 2017-03-13:
sorry - but ipv6 is still an issue... after a reboot 
1. mwan3 is not detecting that the wwan interface is up - only after I run mwan3 restart
2. ipv6 is not working - only after mwan3 stop

feckert, 2017-03-18:
@lollul this issue applies also for the proto ´qmi´ if the option ´dhcp´ is set. I will have a look at this issue.
(@Adze1502 dont want to be the maintainer of this package anymore. So i have added me as a maitainer)

lollul, 2017-03-31:
Checked mwan3_2.4-5. It works great for me with ncm. Thank you!

closed 2017-03-31 by lollul