openwrt/packages #3127: mosquitto version 1.4.10 needs backport of libs/libwebsockets

coelner, 2016-09-03:
In CC the compiler fails if wensockets is activated. With a simple backport the firmware compiles.
```
make[3]: Entering directory '/home/build/openwrt_cc/feeds/packages/net/mosquitto'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16  -I/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/home/build/openwrt_cc/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/build/openwrt_cc/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16  -I/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/home/build/openwrt_cc/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/build/openwrt_cc/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include " LDFLAGS="-L/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/home/build/openwrt_cc/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/build/openwrt_cc/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib " make -j1 -C /home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/. AR="mips-openwrt-linux-uclibc-gcc-ar" AS="mips-openwrt-linux-uclibc-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float" LD=mips-openwrt-linux-uclibc-ld NM="mips-openwrt-linux-uclibc-gcc-nm" CC="mips-openwrt-linux-uclibc-gcc" GCC="mips-openwrt-linux-uclibc-gcc" CXX="mips-openwrt-linux-uclibc-g++" RANLIB="mips-openwrt-linux-uclibc-gcc-ranlib" STRIP=mips-openwrt-linux-uclibc-strip OBJCOPY=mips-openwrt-linux-uclibc-objcopy OBJDUMP=mips-openwrt-linux-uclibc-objdump SIZE=mips-openwrt-linux-uclibc-size CROSS="mips-openwrt-linux-uclibc-" ARCH="mips" WITH_DOCS=no WITH_WEBSOCKETS="yes" ;
make[4]: Entering directory '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10'
set -e; for d in lib client src; do make -C ${d}; done
make[5]: Entering directory '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/lib'
make -C cpp
make[6]: Entering directory '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/lib/cpp'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/lib/cpp'
make[5]: Leaving directory '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/lib'
make[5]: Entering directory '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/client'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/client'
make[5]: Entering directory '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/src'
mips-openwrt-linux-uclibc-gcc mosquitto.o bridge.o conf.o context.o database.o logging.o loop.o memory_mosq.o persist.o net.o net_mosq.o read_handle.o read_handle_client.o read_handle_server.o read_handle_shared.o security.o security_default.o send_client_mosq.o send_mosq.o send_server.o service.o subs.o sys_tree.o time_mosq.o tls_mosq.o util_mosq.o websockets.o will_mosq.o -o mosquitto -L/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/home/build/openwrt_cc/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/build/openwrt_cc/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib  -ldl -lm -lrt -Wl,--dynamic-list=linker.syms -lssl -lcrypto -luuid -lwebsockets
/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib/libwebsockets.so: warning: gethostbyaddr is obsolescent, use getaddrinfo() instead.
/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib/libcrypto.so: warning: gethostbyname is obsolescent, use getnameinfo() instead.
websockets.o: In function `callback_mqtt':
websockets.c:(.text+0x332): undefined reference to `libwebsocket_get_protocol'
collect2: error: ld returned 1 exit status
Makefile:12: recipe for target 'mosquitto' failed
make[5]: *** [mosquitto] Error 1
make[5]: Leaving directory '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/src'
Makefile:21: recipe for target 'mosquitto' failed
make[4]: *** [mosquitto] Error 2
make[4]: Leaving directory '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10'
Makefile:203: recipe for target '/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/.built' failed
make[3]: *** [/home/build/openwrt_cc/build_dir/target-mips_34kc_uClibc-0.9.33.2/mosquitto-ssl/mosquitto-1.4.10/.built] Error 2
make[3]: Leaving directory '/home/build/openwrt_cc/feeds/packages/net/mosquitto'
package/Makefile:191: recipe for target 'package/feeds/packages/mosquitto/compile' failed
make[2]: *** [package/feeds/packages/mosquitto/compile] Error 2
make[2]: Leaving directory '/home/build/openwrt_cc'
package/Makefile:188: recipe for target '/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/build/openwrt_cc/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/build/openwrt_cc'
/home/build/openwrt_cc/include/toplevel.mk:181: die Regel für Ziel „world“ scheiterte
make: *** [world] Fehler 2
```

karlp, 2016-09-03:
Are you sure  you're not running a mixed packages feed?  I'll look at this on monday, but I base most of my builds on CC, I'm not sure how you're seeing this unless you've got somethign else in your tree.

karlp, 2016-09-05:
Thanks for this, it was a typo in a mosquitto change upstream.  I've pushed a patch to the 15.05 branch here, and have filed it upstream as well.  Fixed with 0e43c9c

closed 2016-09-05 by karlp