openwrt/packages #3132: Package php7-mod-mysqli is missing dependencies for the following libraries: libmysqlclient_r.so.16

build000, 2016-09-03:
```
(...)
mkdir -p /home/sj/Pulpit/Projekty/WNDR4700-LEDE/bin/targets/apm821xx/nand/packages /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/php-7.0.10/ipkg-powerpc_464fp/php7-mod-mysqli/CONTROL /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/pkginfo
install -d -m0755 /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/php-7.0.10/ipkg-powerpc_464fp/php7-mod-mysqli/usr/lib/php
install -m0755 /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/php-7.0.10/modules/mysqli.so /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/php-7.0.10/ipkg-powerpc_464fp/php7-mod-mysqli/usr/lib/php/
install -d -m0755 /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/php-7.0.10/ipkg-powerpc_464fp/php7-mod-mysqli/etc/php7
echo "extension=mysqli.so" > /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/php-7.0.10/ipkg-powerpc_464fp/php7-mod-mysqli/etc/php7/mysqli.ini
find /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/php-7.0.10/ipkg-powerpc_464fp/php7-mod-mysqli -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package php7-mod-mysqli is missing dependencies for the following libraries:
libmysqlclient_r.so.16
Makefile:565: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/bin/packages/powerpc_464fp/packages/php7-mod-mysqli_7.0.10-1_powerpc_464fp.ipk' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4700-LEDE/bin/packages/powerpc_464fp/packages/php7-mod-mysqli_7.0.10-1_powerpc_464fp.ipk] Error 1
make[3]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/feeds/packages/lang/php7'
package/Makefile:124: recipe for target 'package/feeds/packages/php7/compile' failed
make[2]: *** [package/feeds/packages/php7/compile] Error 2
make[2]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE'
package/Makefile:121: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE'
/home/sj/Pulpit/Projekty/WNDR4700-LEDE/include/toplevel.mk:193: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
```

closed 2016-09-03 by mhei

build000, 2016-09-04:
Installing the library libmysqlclient_r average for most single-core processors in routers is not a good idea - but in php5 solved this problem without installing libmysqlclent_r.

mhei, 2016-09-04:
Personally, I'm not using mysql(client) on embedded systems, and I did not much think about this. But I agree with your point and will revert this to the old/php5 way.

commit dc56ebbf (2016-09-03): php7: fix two build problems

1) fix rebuilding with existing staging area
2) fix dependency to libmysqlclient (fixes #3132)