openwrt/packages #3142: rng-tools installs dubious default configuration

nwf, 2016-09-05:
Maintainer: @hnyman 
Environment: All

Description:

utils/rng-tools/files/rngd.init defaults to copying data from /dev/urandom back to /dev/random.  While one might have philosophical objections to the kernel's entropy counting mechanisms, this is a complete defeat of those mechanisms and is probably a rude default?  At least, I certainly didn't anticipate that by merely *having* rngd installed I was signing up for that particular stunt.

I'd propose making the file a noop and commenting that this behavior could be enabled if desired?

hnyman, 2016-09-05:
Well, that is how that package operates if you install it. The package is only meant for situations where there is a scarcity of natural entropy and /dev/random needs to be filled with something after a reboot, even if that is not perfect. 6 years ago it was a real problems after Linux devs had removed network drives as an entropy source and Openwrt devs implemented a similar measure here. rng-tools with urandom was introduced at those times. See https://dev.openwrt.org/ticket/9631 for background.

The description might maybe be clarified to mention urandom as a source, but otherwise I am not going to change the package.

Ps. haveged might be a better option, if you need better additional entropy.

nwf, 2016-09-05:
I do run haveged, yes, but thanks for the pointer all the same.

If I UCI-ify the variables in the init script, would you consider taking the patch so that those of us with actual hardware RNGs could point rng-tools at it instead of this dubious (IMHO) faking-out of the kernel entropy counting?

hnyman, 2016-09-05:
Sure, if you want to UCIfy the package and add some configurability to it, that would be great. I would prefer to maintain /dev/urandom as the default source, but there might e.g. be an "enabled" option with a "0" default value in the config file plus the necessary logic in the startup script to prevent "accidental" usage of urandom. 

Naturally you can also manually edit the init script in your own router if you want to use some real entropy source. /etc/init.d/rng-tools in a live router: https://github.com/openwrt/packages/blob/master/utils/rng-tools/files/rngd.init#L8

nwf, 2016-11-01:
Closing now that #3422 is merged.

closed 2016-11-01 by nwf