openwrt/packages #3161: python-pycparser build error [probably latest changes/update python or/and python(3)-setuptools]

build000, 2016-09-14:
make -j8 BUILD_LOG=1

[sources]/logs/package/feeds/packages/python-pycparser/host-compile.txt:

```
Applying ./patches/001-use-external-ply.patch using plaintext: 
patching file pycparser/c_lexer.py
patching file pycparser/c_parser.py
patching file setup.py
running install
running build
running build_py
creating build
creating build/lib
creating build/lib/pycparser
copying pycparser/_build_tables.py -> build/lib/pycparser
copying pycparser/__init__.py -> build/lib/pycparser
copying pycparser/ast_transforms.py -> build/lib/pycparser
copying pycparser/lextab.py -> build/lib/pycparser
copying pycparser/yacctab.py -> build/lib/pycparser
copying pycparser/c_parser.py -> build/lib/pycparser
copying pycparser/c_ast.py -> build/lib/pycparser
copying pycparser/c_lexer.py -> build/lib/pycparser
copying pycparser/_ast_gen.py -> build/lib/pycparser
copying pycparser/plyparser.py -> build/lib/pycparser
copying pycparser/c_generator.py -> build/lib/pycparser
copying pycparser/_c_ast.cfg -> build/lib/pycparser
running install_lib
creating /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/_build_tables.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/__init__.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/ast_transforms.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/_c_ast.cfg -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/lextab.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/yacctab.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/c_parser.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/c_ast.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/c_lexer.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/_ast_gen.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/plyparser.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/c_generator.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/_build_tables.py to _build_tables.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/__init__.py to __init__.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/ast_transforms.py to ast_transforms.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/lextab.py to lextab.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/yacctab.py to yacctab.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/c_parser.py to c_parser.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/c_ast.py to c_ast.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/c_lexer.py to c_lexer.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/_ast_gen.py to _ast_gen.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/plyparser.py to plyparser.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/c_generator.py to c_generator.pyc
running install_egg_info
Writing /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser-2.14-py2.7.egg-info
running install_scripts
running build_scripts
error: cannot copy tree 'build/scripts-2.7': not a directory
Makefile:56: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/pycparser-2.14/.built' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/pycparser-2.14/.built] Error 1
```

hnyman, 2016-09-14:
cc to @jefferyto @commodo 

The error is also picked up by buildbot:
https://downloads.lede-project.org/snapshots/faillogs/mips_24kc/packages/python-pycparser/host-compile.txt

commodo, 2016-09-14:
so, it seems to fail on host-compile
i know about `make package/python-pycparser/compile`
what's the cmd for the host compile ? :)

hnyman, 2016-09-14:
Should be "host-compile" instead of "compile"...

There is a whole bunch of host- commands:
https://git.lede-project.org/?p=source.git;a=blob;f=include/host-build.mk;h=659cdc48374003496d8a7bb2901a93f32952ccc4;hb=HEAD#l191

commodo, 2016-09-14:
hmm ; 

so
```
 sandu@orion  ~/work/sources-work/lede   master  make host-compile V=99
make[1]: Entering directory '/home/sandu/work/sources-work/lede'
make[1]: *** No rule to make target 'host-compile'.  Stop.
make[1]: Leaving directory '/home/sandu/work/sources-work/lede'
/home/sandu/work/sources-work/lede/include/toplevel.mk:193: recipe for target 'host-compile' failed
make: *** [host-compile] Error 2
```

and

```
 sandu@orion  ~/work/sources-work/lede   master  make package/python-pycparser/host-compile V=99
make[1]: Entering directory '/home/sandu/work/sources-work/lede'
make[1]: *** No rule to make target 'package/python-pycparser/host-compile'.  Stop.
make[1]: Leaving directory '/home/sandu/work/sources-work/lede'
/home/sandu/work/sources-work/lede/include/toplevel.mk:193: recipe for target 'package/python-pycparser/host-compile' failed
make: *** [package/python-pycparser/host-compile] Error 2
```

anything i'm missing ?
i tried a few variants before asking the first time :)

commodo, 2016-09-14:
same for `make host/python-pycparser/compile V=99      `

commodo, 2016-09-19:
Managed to build sometimes last week.
It looks like the python-setuptools upgrade broke it.

In the meantime a newer version appeared, that seems to have fixed it (at least on my local build).
Hopefully buildbot will agree too

build000, 2016-09-19:
confirm, actually build without errors:

make -j8 BUILD_LOG=1

[sources]/logs/package/feeds/packages/python-pycparser/host-compile.txt:

```

Applying ./patches/001-use-external-ply.patch using plaintext: 
patching file pycparser/c_lexer.py
patching file pycparser/c_parser.py
patching file setup.py
running install
running build
running build_py
creating build
creating build/lib
creating build/lib/pycparser
copying pycparser/_build_tables.py -> build/lib/pycparser
copying pycparser/__init__.py -> build/lib/pycparser
copying pycparser/ast_transforms.py -> build/lib/pycparser
copying pycparser/lextab.py -> build/lib/pycparser
copying pycparser/yacctab.py -> build/lib/pycparser
copying pycparser/c_parser.py -> build/lib/pycparser
copying pycparser/c_ast.py -> build/lib/pycparser
copying pycparser/c_lexer.py -> build/lib/pycparser
copying pycparser/_ast_gen.py -> build/lib/pycparser
copying pycparser/plyparser.py -> build/lib/pycparser
copying pycparser/c_generator.py -> build/lib/pycparser
copying pycparser/_c_ast.cfg -> build/lib/pycparser
running install_lib
creating /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/_build_tables.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/__init__.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/ast_transforms.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/_c_ast.cfg -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/lextab.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/yacctab.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/c_parser.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/c_ast.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/c_lexer.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/_ast_gen.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/plyparser.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
copying build/lib/pycparser/c_generator.py -> /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/_build_tables.py to _build_tables.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/__init__.py to __init__.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/ast_transforms.py to ast_transforms.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/lextab.py to lextab.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/yacctab.py to yacctab.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/c_parser.py to c_parser.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/c_ast.py to c_ast.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/c_lexer.py to c_lexer.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/_ast_gen.py to _ast_gen.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/plyparser.py to plyparser.pyc
byte-compiling /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser/c_generator.py to c_generator.pyc
running install_egg_info
Writing /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/pycparser-2.14-py2.7.egg-info
Build the lexing/parsing tables
WARNING: yacc table file version is out of date
```
problem fixed with https://github.com/openwrt/packages/pull/3189

closed 2016-09-19 by hnyman