openwrt/packages #3180: luasec: Can't compile after 0.6 update

gendouhydeist, 2016-09-16:
@MikePetullo 

Got this error after compiler applying patch.
>. /home/dkmori/lede/source/include/shell.sh; gzip -dc /home/dkmori/lede/source/dl/luasec-0.6.tar.gz | tar -C /home/dkmori/lede/source/build_dir/target-mips_24kc_musl-1.1.15/luasec-luasec-0.6/.. -xf - 

Applying ./patches/100-luasocket-Makefile.patch using plaintext: 
patching file src/luasocket/Makefile
Hunk #1 FAILED at 6.
Hunk #2 FAILED at 14.
2 out of 2 hunks FAILED -- saving rejects to file src/luasocket/Makefile.rej
Patch failed!  Please fix ./patches/100-luasocket-Makefile.patch!
Makefile:61: recipe for target '/home/dkmori/lede/source/build_dir/target-mips_24kc_musl-1.1.15/luasec-luasec-0.6/.prepared_d17839e5859086e8ba314e7114b10244' failed
make[3]: *** [/home/dkmori/lede/source/build_dir/target-mips_24kc_musl-1.1.15/luasec-luasec-0.6/.prepared_d17839e5859086e8ba314e7114b10244] Error 1
make[3]: Leaving directory '/home/dkmori/lede/source/feeds/packages/lang/luasec'
package/Makefile:124: recipe for target 'package/feeds/packages/luasec/compile' failed
make[2]: *** [package/feeds/packages/luasec/compile] Error 2
make[2]: Leaving directory '/home/dkmori/lede/source'
package/Makefile:121: recipe for target '/home/dkmori/lede/source/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/dkmori/lede/source/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dkmori/lede/source'
/home/dkmori/lede/source/include/toplevel.mk:193: recipe for target 'world' failed

MikePetullo, 2016-09-16:
I am looking at this now.

MikePetullo, 2016-09-16:
Please see:

https://github.com/openwrt/packages/pull/3181

yousong, 2016-10-02:
Should be fixed now.  Thanks for the effort

closed 2016-10-02 by yousong