openwrt/packages #3183: crtmpserver compile error

taeasy, 2016-09-18:
amingclient/src/protocols/m3u8/basem3u8protocol.applestreamingclient.o
../../sources/applications/applestreamingclient/src/clientcontext.cpp: In member function 'bool app_applestreamingclient::ClientContext::ConsumeAVBuffer()':
../../sources/applications/applestreamingclient/src/clientcontext.cpp:389:29: error: 'class InNetTSStream' has no member named 'GetFeedTime'
   _firstFeedTime = pStream->GetFeedTime();
                             ^
../../sources/applications/applestreamingclient/src/clientcontext.cpp:393:44: error: 'class InNetTSStream' has no member named 'GetFeedTime'
  while ((wallClockDelta + 1000 > (pStream->GetFeedTime() - _firstFeedTime)) &&
                                            ^
i486-openwrt-linux-musl-g++ -fPIC -O2  -DLINUX -DLITTLE_ENDIAN_BYTE_ALIGNED -DNET_SELECT -DCreateRTCPPacket=CreateRTCPPacket_mystyle_only_once -DFeedDataAudioMPEG4Generic=FeedDataAudioMPEG4Generic_one_by_one -DHandleTSVideoData=HandleVideoData_version3 -DHAS_PROTOCOL_HTTP -DHAS_PROTOCOL_RTMP -DHAS_PROTOCOL_LIVEFLV -DHAS_PROTOCOL_RTP -DHAS_PROTOCOL_TS -DHAS_PROTOCOL_VAR -DHAS_PROTOCOL_CLI -DHAS_PROTOCOL_HLS -DHAS_PROTOCOL_RAWHTTPSTREAM -DHAS_LUA -DHAS_MEDIA_MP3 -DHAS_MEDIA_MP4 -DHAS_MEDIA_FLV -DHAS_MEDIA_NSV -DHAS_SYSLOG -I../../3rdparty/lua-dev -I../../3rdparty/tinyxml -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I../../sources/common/include -I../../sources/thelib/include -I../../sources/applications/applestreamingclient/include -c ../../sources/applications/applestreamingclient/src/protocols/m3u8/masterm3u8protocol.cpp -o ../../sources/applications/applestreamingclient/src/protocols/m3u8/masterm3u8protocol.applestreamingclient.o
i486-openwrt-linux-musl-g++ -fPIC -O2  -DLINUX -DLITTLE_ENDIAN_BYTE_ALIGNED -DNET_SELECT -DCreateRTCPPacket=CreateRTCPPacket_mystyle_only_once -DFeedDataAudioMPEG4Generic=FeedDataAudioMPEG4Generic_one_by_one -DHandleTSVideoData=HandleVideoData_version3 -DHAS_PROTOCOL_HTTP -DHAS_PROTOCOL_RTMP -DHAS_PROTOCOL_LIVEFLV -DHAS_PROTOCOL_RTP -DHAS_PROTOCOL_TS -DHAS_PROTOCOL_VAR -DHAS_PROTOCOL_CLI -DHAS_PROTOCOL_HLS -DHAS_PROTOCOL_RAWHTTPSTREAM -DHAS_LUA -DHAS_MEDIA_MP3 -DHAS_MEDIA_MP4 -DHAS_MEDIA_FLV -DHAS_MEDIA_NSV -DHAS_SYSLOG -I../../3rdparty/lua-dev -I../../3rdparty/tinyxml -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I../../sources/common/include -I../../sources/thelib/include -I../../sources/applications/applestreamingclient/include -c ../../sources/applications/applestreamingclient/src/protocols/aes/inboundaesprotocol.cpp -o ../../sources/applications/applestreamingclient/src/protocols/aes/inboundaesprotocol.applestreamingclient.o
make[4]: *** [../../sources/applications/applestreamingclient/src/clientcontext.applestreamingclient.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: Leaving directory `/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/crtmpserver-2012-07-18+git-b6fdcdb953d1e99c48a0c37a8c80f2cad2db443b/builders/make'
make[3]: *** [/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/crtmpserver-2012-07-18+git-b6fdcdb953d1e99c48a0c37a8c80f2cad2db443b/.built] Error 2
make[3]: Leaving directory `/home/main/x86/feeds/packages/multimedia/crtmpserver'
make[2]: *** [package/feeds/packages/crtmpserver/compile] Error 2
make[2]: Leaving directory `/home/main/x86'
make[1]: *** [/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/x86'

yousong, 2016-11-09:
Can you confirm pr https://github.com/openwrt/packages/pull/3418 fixes this issue for you?

diizzyy, 2017-06-02:
@hnyman @thess 
crtmpserver compiles, please close this issue report.

closed 2017-06-02 by hnyman