openwrt/packages #3191: logrotate build error - missing dependencies libacl.so.1

build000, 2016-09-19:
make -j1 BUILD_LOGS=1
[sources]/logs/package/feeds/packages/logrotate/compile.txt
```
(...)
find /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/logrotate-3.10.0/ipkg-powerpc_464fp/logrotate -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package logrotate is missing dependencies for the following libraries:
libacl.so.1
Makefile:60: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/bin/packages/powerpc_464fp/packages/logrotate_3.10.0-1_powerpc_464fp.ipk' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4700-LEDE/bin/packages/powerpc_464fp/packages/logrotate_3.10.0-1_powerpc_464fp.ipk] Error 1
make[3]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/feeds/packages/utils/logrotate'
```

- of course this dependencies libs build properly:

[sources]/logs/package/feeds/packages/acl/compile.txt:
```
home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/root-apm821xx/tmp-libacl/usr/lib/
SHELL= flock /home/sj/Pulpit/Projekty/WNDR4700-LEDE/tmp/.root-copy.flock -c 'cp -fpR /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/root-apm821xx/tmp-libacl/. /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/root-apm821xx/'
rm -rf /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/root-apm821xx/tmp-libacl
touch /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/root-apm821xx/stamp/.libacl_installed
make[3]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/feeds/packages/utils/acl'
```

hnyman, 2016-09-19:
Buildbot is not picking that up and the package is available also for your platform:
https://downloads.lede-project.org/snapshots/packages/powerpc_464fp/packages/logrotate_3.10.0-1_powerpc_464fp.ipk

Is there anything special in your build?

cc to the package maintainer: @bk138

build000, 2016-09-19:
no it is not anything special - just that originally was building the -j8 also discovered that the python-cffi not built a logrotate may falsely indicate that it is not built, although the built de facto - a typical ailment cross-compile ...
In general, I do not have confidence, so I reported first on what stopped make -j1.

bk138, 2016-09-19:
It's probably that libacl is installed on the build machine which logrotate's configure is picking up. There probably is a configure switch to disable looking for ACL, but by looking at logrotate's configure.ac, there's more cases like that, e.g. with SELinux.

What's the general approach here: assuming a minimal build env or disabling all might-be deps specifically?

build000, 2016-09-19:
libacl and logrotate build only to [m] - also I do not know exactly how it is recognized by the system builder.

hnyman, 2016-09-19:
> What's the general approach here: assuming a minimal build env or disabling all might-be deps specifically?

Typical approach is to disable might-be deps with CONFIGURE_ARGS in Makefile if the package's configure script has such a parameter, or alternatively via patching the config script. Build environments vary and we can't assume a minimal build.

build000, 2016-09-19:
Overall, it surprised me too this error message because logrotate is dependent libpopt and not from libacl - as usual out some incredible Kaszana logs at multi-threaded compilation - still LEDE / OpenWRT is not in this respect by the end polished.

closed 2016-09-19 by build000