openwrt/packages #3192: python-cffi build error

build000, 2016-09-19:
make -j1 V=s
```
(...)
make[3]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/feeds/packages/lang/python-cffi'
if [ "HOST" == "HOST" ]; then export PYTHONPATH="/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7:/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host//lib/python2.7/site-packages"; export PYTHONDONTWRITEBYTECODE=0; else export PYTHONPATH="/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/lib/python2.7:/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15//usr/lib/python2.7/site-packages:/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/cffi-1.7.0/ipkg-install//usr/lib/python2.7/site-packages"; export PYTHONDONTWRITEBYTECODE=1; export _python_sysroot="/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15"; export _python_prefix="/usr"; export _python_exec_prefix="/usr"; fi; export PYTHONOPTIMIZE="";  cd /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/cffi-1.7.0/; CC="gcc" CCSHARED="gcc -fPIC" CXX="g++" LD="gcc" LDSHARED="gcc -shared" CFLAGS="-O2 -I/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/include -I/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/usr/include -I/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include" CPPFLAGS="-I/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/include -I/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/usr/include -I/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include -I/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include/python2.7" LDFLAGS="-L/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/lib -L/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/usr/lib -L/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib -lpython2.7 -Wl,-rpath=/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib" _PYTHON_HOST_PLATFORM=linux2   /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/bin/python2.7  ./setup.py install --prefix="" --root="/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host" ;
Traceback (most recent call last):
  File "./setup.py", line 118, in <module>
    from setuptools import setup, Distribution, Extension
  File "/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/setuptools/__init__.py", line 14, in <module>
    from setuptools.extension import Extension
  File "/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/setuptools/extension.py", line 10, in <module>
    from .dist import _get_unpatched
  File "/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/setuptools/dist.py", line 20, in <module>
    from setuptools.monkey import get_unpatched
  File "/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/lib/python2.7/site-packages/setuptools/monkey.py", line 10, in <module>
    from .py26compat import import_module
ImportError: cannot import name import_module
Makefile:54: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/cffi-1.7.0/.built' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/cffi-1.7.0/.built] Error 1
make[3]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/feeds/packages/lang/python-cffi'
package/Makefile:124: recipe for target 'package/feeds/packages/python-cffi/host/compile' failed
make[2]: *** [package/feeds/packages/python-cffi/host/compile] Error 2
make[2]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE'
package/Makefile:121: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE'
/home/sj/Pulpit/Projekty/WNDR4700-LEDE/include/toplevel.mk:193: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
```
maybe upgrade to a newer version of python-setuptools requires more than just fixes : https://github.com/openwrt/packages/issues/3161

hnyman, 2016-09-19:
That error is picked up by the buildbot, too. E.g. https://downloads.lede-project.org/snapshots/faillogs/mipsel_74kc/packages/python-cffi/host-compile.txt

cc @commodo @jefferyto

commodo, 2016-09-19:
pfff.....
so, one host compile fixed, and then moves to another host compile...

oh, well ; will check :)

jefferyto, 2016-09-20:
Even with `python-cffi` updated to the latest version, this error persists, but I think this is related to the python installation process itself.

After trying to compile `python-cryptography` (which depends on `python-cffi/host`) and hitting this error, I looked into staging_dir:
* In `staging_dir/target-xxx/usr/lib/python2.7/site-packages` (note: non-host), there are `setuptools` and `setuptools-20.10.1.dist-info` directories. Note the version: 20.10.1
* In `staging_dir/target-xxx/host/lib/python2.7/site-packages`, there are `setuptools`, `setuptools-20.10.1.dist-info` and `setuptools-27.2.0-py2.7.egg-info` directories. The files in `setuptools` appear to be a combination of the two versions.

Looking at the timestamps, it seems 20.10.1 was installed before 27.2.0, and it seems 20.10.1 was installed at the same time as pip 8.1.1 (pip also exists in two `site-packages` directories from above). But `python-pip` (the OpenWrt package) wasn't installed or required by any other package in this process.

Going through the make output, there are references to `ensurepip` in the python build process. And looking inside Python-2.7.12.tar.xz, in the `Lib/ensurepip/_bundled` directory are `setuptools-20.10.1-py2.py3-none-any.whl` and `pip-8.1.1-py2.py3-none-any.whl`.

So I think python is installing it's own pip and setuptools, which is conflicting with the newer setuptools, and is causing the build error (i.e. `py26compat.py` in the older setuptools didn't have code involving `import_module`).

Maybe @commodo or someone else more familiar with the python install process can take a look to verify this?

build000, 2016-09-20:
So how ? To be able to use newer versions of python-setuptools and python-pip need to strip the current version of the python with its built-pip and stuptools? I think there's a more civilized way to reconcile references to existing and non-existing versions of packages /libraries/scripts?

commodo, 2016-09-21:
yep; python & python3 are installing it's own pip & setuptools but only for the host-side build
i forgot about this ; i only remembered while re-looking into the Makefiles

@jefferyto did you try to see if that setuptools/pip is sufficient ? maybe we could disable the python-setuptools host-side build ; not sure

atm, i am trying to disable python's setuptools 

maybe long-term we could drop setuptools & pip and just install python's (for both host-side & target-side)
i am not sure if ever tried this ; i may give it another try real quick
since i maintain all 3 packages (python, python-setuptools & python-pip & also the python3 variants), it would sound a bit easier to try to merge them

commodo, 2016-09-21:
in the meantime: https://github.com/openwrt/packages/pull/3201

build000, 2016-09-21:
confirm - this patch fixed all problems build python-cffi and all other dependencies https://github.com/openwrt/packages/commit/a4b0c0a9fb4a767f1e7fe344423f77ef1f10d99
Issue to by closed.

closed 2016-09-21 by build000