openwrt/packages #3209: l7-protocols still broken?

CanadianJeff, 2016-09-22:
```
root@WDR4300-JULES:~# opkg install luci-app-p2pblock
Installing luci-app-p2pblock (git-15.248.30277-3836b45-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packag                                                                          es/luci/luci-app-p2pblock_git-15.248.30277-3836b45-1_all.ipk.
Installing freifunk-p2pblock (3) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packag                                                                               es/luci/freifunk-p2pblock_3_ar71xx.ipk.
Installing iptables-mod-ipp2p (2.5-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packag                                                                               es/base/iptables-mod-ipp2p_2.5-1_ar71xx.ipk.
Installing kmod-ipt-ipp2p (3.18.20+2.5-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packag                                                                               es/base/kmod-ipt-ipp2p_3.18.20+2.5-1_ar71xx.ipk.
Installing kmod-ipt-compat-xtables (3.18.20+2.5-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packag                                                                               es/base/kmod-ipt-compat-xtables_3.18.20+2.5-1_ar71xx.ipk.
Installing l7-protocols (2009-05-28-2) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packag                                                                               es/packages/l7-protocols_2009-05-28-2_ar71xx.ipk.
Configuring kmod-ipt-compat-xtables.
Configuring kmod-ipt-ipp2p.
Configuring iptables-mod-ipp2p.
Configuring l7-protocols.
Configuring freifunk-p2pblock.
freifunk-p2pblock: starting p2pblock...
iptables v1.4.21: Couldn't load match `layer7':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.4.21: Couldn't load match `layer7':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.4.21: Couldn't load match `layer7':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.4.21: Couldn't load match `layer7':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.4.21: Couldn't load match `layer7':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.4.21: Couldn't load match `layer7':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
freifunk-p2pblock: Done.
Configuring luci-app-p2pblock.
```

hnyman, 2016-09-22:
I think that l7-protocols package itself may work (as it just installs the protocol definitions), but the actual l7 support was removed from the firewall with https://dev.openwrt.org/changeset/45424 and https://dev.openwrt.org/changeset/45423 in early 2015. (Those commits affect both trunk and CC15.05 that was branched later.)

So, the freifunk-p2pblock should probably be modified (or declared broken).

CanadianJeff, 2016-09-22:
speaking with the clearos team they have something interesting???

https://www.clearos.com/clearfoundation/social/community/two-new-apps-application-filter-and-protocol-filter

might be worth looking into it

CanadianJeff, 2016-09-22:
also that page links to something else of interest

http://www.ntop.org/products/deep-packet-inspection/ndpi/

hnyman, 2017-05-08:
closing, as this is old and there is no change in l7 support.

closed 2017-05-08 by hnyman

Yannis100, 2017-05-23:
L7-protocols itself is dead but nDPI is updated regulalry, what about adding support for it ?
https://github.com/betolj/ndpi-netfilter
https://github.com/ntop/nDPI
#4049