openwrt/packages #3235: prosody 0.9.10: ssl not working with luasec 0.6

mgondium, 2016-09-26:
Maintainer: heil@terminal-consulting.de
Environment: LEDE head on ath9k (TP-LINK 1024ND v1)

Description:
"Unable to initialize TLS: LuaSec (required for encryption) was not found" on prosody.err

Prosody 0.9.10 requires luasec for SSL, but luasec 0.6 does not support Lua 5.1:
=> https://github.com/voidlinux/void-packages/issues/3778

From LEDE head on ath9k:
```
 @prosodyctl about
 Prosody 0.9.10
 # Lua environment
 Lua version:             	Lua 5.1
 # Lua module versions
 lfs:     	LuaFileSystem 1.6.2
 lxp:     	LuaExpat 1.3.0
 pposix:  	0.3.6
 socket:  	LuaSocket 3.0-rc1
 ssl:     	0.6
```

MikePetullo, 2016-10-15:
I have asked about updating the lua to 5.2+ package both in OpenWrt's GitHub issue tracker and LEDE's mailing list. See:

    https://github.com/openwrt/openwrt/issues/136

and:

    http://lists.infradead.org/pipermail/lede-dev/2016-October/003361.html

diizzyy, 2016-10-16:
@MikePetullo 
http://lists.infradead.org/pipermail/lede-dev/2016-October/003357.html might be of interest
Feel free to pick it up

wcat13, 2016-12-04:
Have you guys had any luck working this out? I cannot get Prosody up and running. 

```
root@chat:~# prosodyctl about
Prosody 0.9.10

# Prosody directories
Data directory:  	/etc/prosody/data
Plugin directory:	/usr/lib/prosody/modules/
Config directory:	/etc/prosody
Source directory:	/usr/lib/prosody

# Lua environment
Lua version:             	Lua 5.1

Lua module search paths:
  /usr/lib/prosody/?.lua
  /usr/share/lua/?.lua
  /usr/share/lua/?/init.lua
  /usr/lib/lua/?.lua
  /usr/lib/lua/?/init.lua

Lua C module search paths:
  /usr/lib/prosody/?.so
  /usr/lib/lua/?.so
  /usr/lib/lua/loadall.so

LuaRocks:        	Not installed

# Lua module versions
lfs:     	LuaFileSystem 1.6.2
lxp:     	LuaExpat 1.3.0
pposix:  	0.3.6
socket:  	LuaSocket 3.0-rc1
ssl:     	0.6
```

This is LEDE trunk and have the same TLS error. I found mention of package lua51-luasec
https://github.com/voidlinux/void-packages/issues/3778

But do not know where to obtain it.

vcunat, 2017-05-01:
Is this still a problem?  Prosody was updated to 0.9.12 and that version is supposed to be fixed with luasec 0.6: http://blog.prosody.im/prosody-0-9-12-released/

mgondium, 2017-05-01:
Just tested on LEDE and prosody 0.9.12 is working with SSL, even though prosody.err shows

```
May 01 14:35:11 portmanager	error	Failed to open server port 5222 on ::, Address family not supported by protocol
May 01 14:35:11 portmanager	error	Failed to open server port 5269 on ::, Address family not supported by protocol
May 01 14:35:11 localhost:tls	error	Unable to initialize TLS: No SSL/TLS configuration present for localhost
May 01 14:35:11 localhost:tls	error	Unable to initialize TLS: No SSL/TLS configuration present for localhost
```
This may be caused by the lack of IPv6 on my particular configuration.
Nonetheless, it works after generating a new key and certificate (pidgin + encryption required).
Cheers.

wcat13, 2017-05-01:
I ended up throwing prosody on one of those little $9 chip minicomputers.  With prosody-daily or whatever that package is called.  Nice to know that this lede package is being maintained and used by others though.

neheb, 2019-03-30:
latest master has a newer version. closing.

closed 2019-03-30 by neheb