openwrt/packages #3239: boost not building on build servers

Be-ing, 2016-09-26:
Maintainer: @ClaymorePT
Environment: all

I tried to install mpd from http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/ but it was mysteriously not there. I asked about this in #lede-dev and was pointed to the build logs http://downloads.lede-project.org/snapshots/faillogs/arm_cortex-a9_vfpv3/packages/ which do not mention mpd. neoraider identified that the problem seems to be with boost, which mpd depends on:

[18:26] <neoraider> boost has DEPENDS:=+ALL:boost-libs
[18:26] <neoraider> So when CONFIG_ALL is set (all packages are built, like on the build bots), boost depends on all boost libraries???
[18:26] <neoraider> And one of the libraries depends on CONFIG_BUILD_NLS, which is not set in the default config, and thus not on the build bots
[18:27] <neoraider> I'd recommend opening an issue against the boost package

ClaymorePT, 2016-09-27:
Thanks for the warning. 
Because changing configuration options in the kernel is something not very welcomed by the OpenWRT Maintainers, the automatic activation of CONFIG_BUILD_NLS upon selection of Boost.Locale was removed in commit https://github.com/openwrt/packages/commit/599f289585e61db72ab1d79404f507cc28e9d6ee

There is a warning in the help section of the Boost Libs.

Try copying the Boost Package from OpenWRT to LEDE packages directory and activate the full language support in the LEDE kernel. That should be enough to build Boost 1.61

Also, Boost 1.62 will be released tomorrow (28th of september) and because of this, I will submit two new packages releases, one for OpenWRT and another one for LEDE. If the LEDE maintainers allow it, I will re-add +@BUILD_NLS for Boost.Locale once again (which I think its the correct approach).

thess, 2016-09-27:
I know that build dependencies can be a bit complicated when we are sub-setting packages but I don't think it best to have the tail wag the dog here. BUILD_NLS is a global setting and affects many packages. boost.locale should not build unless BUILD_NLS is selected explicitly. @nbd168 's change was correct.

That said, I also don't think the selection of one boost library as a dependency in a package (MPD) should force the installation (and compilation) of all of boost. It would also be convenient if all boost libraries could be installed without having to enumerate them manually for opkg commands.

So, given these goals, I believe a full build of all packages without having BUILD_NLS set should produce the correct results. boost.locale should not be built and all other libraries (including the all-inclusive -libs package) be correctly built and individual package dependencies be met.

We should find a way to fix this and not have optional libraries forcing global build options.

ClaymorePT, 2016-09-27:
And that is why I did not complain about the change at the time. :)

The thing is, Boost Libs are in their majority "include only" libs.
Only Boost.Locale requires iconv with a kernel compiled with full language
support.

The only solutions I could think of are:
1) Select none of the "compiled libraries" and copy "header only" libraries
2) Select all of the libraries except Boost.Locale

Opinions?

thess, 2016-09-28:
I made an attempt to solve these build issues - check out #3254

ClaymorePT, 2016-09-29:
I have opened a pull request - https://github.com/openwrt/packages/pull/3265

thess, 2016-10-08:
Original issue is resolved with updated 1.61.0 version of Boost. Closing this

closed 2016-10-08 by thess